OpenEtherCATsociety/SOEM #103: Control CoE Device

AndyElogic, 2017-04-27:
I have a slave using the CoE protocol,And I have a list of objects,But I don't know how to use these objects to send command to my Slave.

Thank you

nakarlsson, 2017-04-27:
https://github.com/OpenEtherCATsociety/SOEM/issues/96 mention and links further on howto.

AndyElogic, 2017-04-28:
Thank you for your reply, But I don't understand what the following part of the program is used to do

 +/* Explicitly set flags that are (probably) invalid in EEPROM */
 +   ec_slave[slave].SM[2].SMflags = 0x10024;
 +
 +   /* Explicitly disable sync managers that are activated by EEPROM */
 +   ec_slave[slave].SM[4].StartAddr = 0;
 +   ec_slave[slave].SM[5].StartAddr = 0;

nakarlsson, 2017-04-28:
That's a special bugfix for that device, you can ignore that. Just grab the SDO logic

AndyElogic, 2017-04-28:
About the following part of the program,I don't understand what the "PO2SOconfig" parameter means, And why "moog_setup" don't have parameter

+               if(slave->eep_man == 0x28 && slave->eep_id == 0x1E0)
 +               {
 +                  printf ("Found Moog drive\n");
 +                  slave->PO2SOconfig = moog_setup;
 +               }

mheden, 2017-04-28:
`PO2SOconfig` is for registering a hook function that will be called when the slave does the transition between Pre-OP and Safe-OP.

AndyElogic, 2017-05-04:
I have mimicked the program
I try to set the object 0x6040(ControlWord) via moog_write16(slave, 0x6040, 0, 0x06),
But I read the object 0x6041(Statusword) via moog_read(slave, 0x6041, 0, value),The Value always is "24",

The moog_read() Define :
static int moog_read(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
	int wkc;
	int l = sizeof(value);

	wkc = ec_SDOread(slave, index, subindex, FALSE, &l, &value, EC_TIMEOUTRXM);

	return wkc;
}

Did i set the object in such a way?

Thank you

nakarlsson, 2017-05-04:
6041 is a Status Word, set by the slave.

AndyElogic, 2017-05-05:
Is not by set "control word" after that can read "status word" monitor the state?
I am not familiar with CANopen,sorry..

nakarlsson, 2017-05-05:
What it means is defined by the CiA402, I'd think you'll find it valuable to download that.

AndyElogic, 2017-05-09:
I want make my Slave "servo on", Through
mooog_write16(slave, 6040, 0, 0x06) ->"Ready to switch on"
mooog_write16(slave, 6040, 0, 0x07) ->"Switch on"
mooog_write16(slave, 6040, 0, 0x0F) ->"Operation enable"
but why the state machine is still in state "Switch on"?

And whether there are examples of use on ec_RxPDO and ec_TxPDO

Thank you

nakarlsson, 2017-05-09:
1. You should study the Commissioning chapter in your drive reference manual.

2. For EtherCAT you don't use ec_RxPDO/ec_TQxPDO, instead you configure and map you slaves to the logic addressspace and use LRW via ec_send_processdata/ ec_receive_processdata . Check simpletest for example

nakarlsson, 2017-05-20:
No update so fixed I guess

closed 2017-05-20 by nakarlsson

NTESEC-zg, 2018-09-24:
how is the mechanism in SOEM that can make my motor(driver) move after the configuration and mapping.