OpenEtherCATsociety/SOEM #105: how to configure Distributed Clocks?

LuciusMagnus, 2017-05-09:
I'm don't know much about controlling EtherCAT slaves.
I'm using SOEM library in order to control a beckhoff servo motor (AM8131-0F20-0000) which is connected to the servo motor terminal EL7211-0010 through an OCT cable. The EL7211-0010 is connected to the EK1100 coupler terminal.

i added a single line code "ec_dcsync0(1, TRUE, 100000U, 0)" to simpletest function in the simple_test.c file to configure the distributed clocks. When I run the program, it says "Invalid DC SYNC configuration". I've read some materials online, and it's said that the reference clock and the local clocks need to be configured so that the process data transferred between master (in my case, the linux PC) and slaves gets synchronized. But I don't know how to do it. So, I'd be very thankful if someone could help configure the Distributed Clocks to get the servo motor running, please.
`
void simpletest(char *ifname)
{
      ....
    if (ec_init(ifname))  // initialize SOEM, bind socket to ifname 
    {   
           printf("ec_init on %s succeeded.\n",ifname);
           if ( ec_config_init(FALSE) > 0 )   //find and auto-config slaves 
           {
                 printf("%d slaves found and configured.\n",ec_slavecount);
                 ec_config_map(&IOmap); //IOmap configuration

        	 //Distributed clocks configuration
                 ec_configdc(); 
         	 ec_dcsync0(1, TRUE, 100000U, 0);`
    ....
}

nakarlsson, 2017-05-09:
If you read #29, you'll find a ref to a post on TI forum by Arthur, it explains the steps to do and also a better example, test/Linux/red_test.c

nakarlsson, 2017-05-20:
No update so fixed I guess, #29 got it all

closed 2017-05-20 by nakarlsson

wschoenell, 2019-04-24:
As a reference for the future readers of this thread: I could control the same servo with the EL7211 by modifying the red_test.c code as in wschoenell/SOEM@689a947