OpenEtherCATsociety/SOEM #108: Add official support for Xenomai

zchen24, 2017-05-27:
I was wondering is there is a plan to add official support for Xeonmai + RTNet.

nakarlsson, 2017-05-27:
No, most of what you see in todays SOEM are based on customer driven development. So if s customer project would require support for Xenomai we would add it.
That said I'm sure there already exist ports out there.

closed 2017-05-31 by nakarlsson

zchen24, 2017-06-01:
Hi Andreas, 

If I have a "port" that works on Xenomai and would like to contribute back, are you open to this? 

Thanks,
Zihan

nakarlsson, 2017-06-02:
yes

First step is to sign Contributor License Agreement: https://openethercatsociety.github.io/cla/cla_soem_soes.pdf

Second step is to create a pull request for review

zchen24, 2017-06-03:
Thanks for the information.

zchen24, 2017-06-21:
Just sent the signed Contributor License Agreement to openethercatsociety@rt-labs.com.
Will send a pull request late next month. 

Thanks,
Zihan

zchen24, 2017-07-12:
Just sent a pull request. 

Thank you,
Zihan

roboticsai, 2019-08-28:
where can i find the soem integrated with rtnet and xenomai? if it is not their, how can i do it myself?