OpenEtherCATsociety/SOEM #111: cannot receive processdata

joeization, 2017-05-31:
when trying simple test
the output is like this

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on something succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 96 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
-1
OK : all slaves resumed OPERATIONAL.
-1
OK : all slaves resumed OPERATIONAL.
-1

I add the output for ec_receive_processdata's wkc, -1 in my case

why the slave is configured and i cant receive pdo?

nakarlsson, 2017-05-31:
What plattform do you run? Do you have an wireshark?

From the current information I seems you timeout and don't receive the frame within expected timeout.
What happens if you call ec_receive_processdata with longer timeout?

joeization, 2017-05-31:
plattform is windows 7 and i'll try wireshark later
ec_receive_processdata(10000) and output still the same

joeization, 2017-05-31:
here is the wireshark output

Frame 1168: 30 bytes on wire (240 bits), 30 bytes captured (240 bits) on interface 0
Ethernet II, Src: Private_01:01:01 (01:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 'BRD': Len: 2, Adp 0x0, Ado 0x130, Wc 0 

Frame 1169: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface 0
Ethernet II, Src: 03:01:01:01:01:01 (03:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 'BRD': Len: 2, Adp 0x1, Ado 0x130, Wc 1 
Pad bytes: 000000000000000000000000000000000000000000000000...

Frame 1170: 144 bytes on wire (1152 bits), 144 bytes captured (1152 bits) on interface 0
Ethernet II, Src: Private_01:01:01 (01:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 2 Cmds, 'LRW': len 96, 'FRMW': len 8

nakarlsson, 2017-05-31:
The interesting frame would be Frame 1171 , the response of the LRW, what its Wc is.

joeization, 2017-05-31:
sorrry i restart wireshark and get this

Frame 1061: 144 bytes on wire (1152 bits), 144 bytes captured (1152 bits) on interface 0
Ethernet II, Src: Private_01:01:01 (01:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 2 Cmds, 'LRW': len 96, 'FRMW': len 8 

Frame 1062: 30 bytes on wire (240 bits), 30 bytes captured (240 bits) on interface 0
Ethernet II, Src: Private_01:01:01 (01:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 'BRD': Len: 2, Adp 0x0, Ado 0x130, Wc 0 

Frame 1063: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface 0
Ethernet II, Src: 03:01:01:01:01:01 (03:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 'BRD': Len: 2, Adp 0x1, Ado 0x130, Wc 1 
Pad bytes: 000000000000000000000000000000000000000000000000...

Frame 1064: 144 bytes on wire (1152 bits), 144 bytes captured (1152 bits) on interface 0
Ethernet II, Src: Private_01:01:01 (01:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 2 Cmds, 'LRW': len 96, 'FRMW': len 8

nakarlsson, 2017-05-31:
So from looking at the wireshark log provided,  you don't get a response LRW frame and receive return -1 correctly. You need to investigate why the LRW get thrown away.

joeization, 2017-05-31:
What should i do?

nakarlsson, 2017-05-31:
Do you take the wireshark log by sniffing the EtherCAT master NIC on the computer OR from an EtherCAT Master NIC -> switch  -> EtherCAT Slave  and sniff from the switch?  

Try from a Linux box?

Turn off firewalls on your windows box?

Switch NIC on your windows box? Try not using a USB dongle NIC

joeization, 2017-06-01:
I've tried all of them except the linux box.
but when i use TwinCAT, it does recive and send PDO
here is the wireshark log

Frame 6710: 79 bytes on wire (632 bits), 79 bytes captured (632 bits) on interface 0
Ethernet II, Src: Beckhoff_01:00:00 (01:01:05:01:00:00), Dst: MS-NLB-PhysServer-16_f3:5a:65:b4 (02:10:f3:5a:65:b4)
EtherCAT frame header
EtherCAT datagram(s): 3 Cmds, 'LRD': len 1, 'LRW': len 24, 'BRD': len 2

Frame 6711: 79 bytes on wire (632 bits), 79 bytes captured (632 bits) on interface 0
Ethernet II, Src: NexcomIn_5a:65:b4 (00:10:f3:5a:65:b4), Dst: Beckhoff_01:00:00 (01:01:05:01:00:00)
EtherCAT frame header
EtherCAT datagram(s): 3 Cmds, 'LRD': len 1, 'LRW': len 24, 'BRD': len 2

nakarlsson, 2017-06-01:
If you use this setup with SOEM "Master NIC -> switch -> EtherCAT Slave" and sniff with another NIC/USB NIC. Do you see the PDO send and receive in wireshark?

joeization, 2017-06-03:
Is it possible some how soem send wrong payload?

nakarlsson, 2017-06-03:
Nothing is impossoble, it's for you to provide relevant info to be able to solve the problem.  So far its hard to make any conclosions.

Try the setup with switch and sniff that and try linux.

joeization, 2017-06-04:
sorry i'm near the end of the semester and have lot of project to do
i will dig into this in july

ArthurKetels, 2017-06-06:
I guess you have a slave with no LRW capability. Most likely a TI Sitara processor. It also means its eeprom is wrong in that it does not set the LRD/LWR only capability. Forcing SOEM to use LRD/LRW will solve this.

joeization, 2017-06-08:
Is there any example showing how to set it?

mheden, 2017-06-08:
Set 
```
ec_slave[<slave_num>].blockLRW = 1;
```
for you slave between the calls to `ec_config_init` and `ec_config_map`

joeization, 2017-07-18:
> ec_slave[<slave_num>].blockLRW = 1;

it works

now i can send and receive pdo

thanks!

nakarlsson, 2017-07-18:
Ok, we'll close the issue then

closed 2017-07-18 by nakarlsson