OpenEtherCATsociety/SOEM #114: Can not change state of slave after changing PDO Map

dan-choe, 2017-07-06:
Hello,

I want to test response time of each PDO, so that I changed PDO Map on Pre-OP state.
I only use one slave.  
After changing PDO Map, I can't change state of slave from Pre-OP to OP even SAFE-OP.
The state of slave is changed to EC_STATE_PRE_OP_ERR.
Do you know why I can't change the state of slave in this situation?

closed 2017-07-06 by dan-choe

nakarlsson, 2017-07-07:
Nope, I think you need to consult the manual, something isn't configured correct preventing you from entering safe-op & op.

dan-choe, 2017-07-07:
When one of TX and RX is empty, I couldn't change the state of slave from Pre-OP to OP.
Is there any idea how to change state to OP if one of them is empty?

nakarlsson, 2017-07-08:
If all refeences is empty, make sure you clear the SM object read from the EEPROM.

dan-choe, 2017-07-11:
hum... I update SM assignment after removing all objects of one of 0x1A00 and 0x1600.
I still can't change Pre-OP to OP. How do I check SM object read from the EEPROM is cleared?
Sorry, I'm new to EtherCAT and SOEM.

nakarlsson, 2017-07-11:
Run slaveinfo and and switch -map and -sdo and post the result. There you see how SOEM interpret the slave Online data including EEPROM data.