OpenEtherCATsociety/SOEM #118: Updating SyncManager Assignment for Multi Slaves

dan-choe, 2017-07-19:
Hello,

When I use default map 0x1600 and 0x1a00 after updating Updating SyncManager Assignment, PDOs in default map works well on both slaves.
However, if I update SyncManager Assignment to 0x1601 and 0x1A01, PDOs do not work on first slave, but second slave. 
There is no error, just no response from first slave.

Do you have any idea?
Thank you for your time.

nakarlsson, 2017-07-19:
Do you get correct wkc?

dan-choe, 2017-07-19:
um. I'm checking wireshark to find out what's different between 0x1600 and 0x1601.
When I write value of PDO, The state is Operational state. State is stable. It is not changed while keep reading and writing PDOs.
wkc is 6. that means rw type object write value two times correctly.

nakarlsson, 2017-07-19:
You shoud do change in preop.

dan-choe, 2017-07-19:
ah. when I update Syncmanager assignment to either 0x1600 or 0x1601, the state is Pre-OP.
I set precodition in Pre-OP, then I use OP state to do cyclical data exchange( reading & writing PDO).

nakarlsson, 2017-07-19:
I remember a smilar case that required DC to be active for the slave to update the PDOs.

The behaviour of that slave, entering OP without error and not updating the PDOs seems faulty to me though.

dan-choe, 2017-07-19:
um. after updating SyncManager Assignment to 0x1601 and 0x1A01.
I print out all the mapped object entries. They print out correct objects of 0x1601 and 0x1A01.
i don't know why first slave does not work with PDO while second slave works well.

nakarlsson, 2017-07-19:
If you read the PDO value via SDO, can you confirm that the slave value return is wrong?

dan-choe, 2017-07-19:
I test it via SDO for 0x1600 and 0x1601. They both returns right values. 

What I want to do with slaves is that turning ON the device. 
0 indicates OFF. 1 indicates ON.

Before requesting, the value is 0 as OFF.
so if I use 0x1600 and request to turn on. It returns 1.  That means the device is turned ON.

But, if I use 0x1601 and request to turn on for both slaves. 
First slave returns 0.  Second slave returns 1.  That means the first device is not turned on while second device is turned ON.

Both slaves have same PDO index number.

nakarlsson, 2017-07-19:
I think you should consult the slave manufacturer

dan-choe, 2017-07-19:
um.. Okay! Thanks for all your time! 👍 

I'm majoring Computer Science and an undergraduate student. 
I didn't learn about Hardware, Hardware Networking at school.
What do I have to study to make SOEM like you? Programming is not issue. 
I want to study about EtherCAT and SOEM. cuz I want to control EtherCAT whatever I want to. and I want to use Ethercat with Aduino in the future.

nakarlsson, 2017-07-19:
I this particular case you need to understand how the slaves work/should work. If I understand you correct, SOEM is working as expected while the slave don't,

dan-choe, 2017-07-19:
Hum.. I will keep studying about it. Thanks.
and the problem is solved. Lol
I think the given 0x1601 map has problem. I don't know what exactly point has an issue.
I just made new 0x1601 map, and both slaves work well.

I really appreciate your help and you provides SOEM. :D 👍

closed 2017-07-19 by dan-choe