OpenEtherCATsociety/SOEM #119: Mapping the same memory area

stefanpoeter, 2017-07-20:
slaveinfo outputs the following

```
Slave:11
 Name:EL2624
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 1425[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 100b
 Man: 00000002 ID: 0a403052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:3 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 200[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1602 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1603 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output

Slave:12
 Name:EL2624
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 1570[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 100c
 Man: 00000002 ID: 0a403052 Rev: 00120000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:4 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 200[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1602 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1603 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output
```

The bits mapped in memory are covering the same memory area. All of them are on the first four bits from the first byte. I would expect it to be 0x0000.0-0x0000.3 for slave 11 and 0x0001.0-0x0001.3 for slave 12. Am I missing something here?

mheden, 2017-07-20:
SOEM maps bit slaves to the same byte if there are space available so in
this case they will be at the same byte (slave 1 0x0.0-0x0.3, slave 2,
0x0.4-0x0.7).

However, I think slaveinfo has a bug that doesn't calculate the bit offset
correctly in this kind of system.

nakarlsson, 2017-07-20:
You can see the actual FMMU config in the following lines, as mheden mention, they share byte and 11 start at bit 0 and end at bit 3, 12 start at bit 4 and end at 7.
Slave 11,
"FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:3"

Slave 12, 
"FMMU0 Ls:00000000 Ll:   1 Lsb:4 Leb:7"

stefanpoeter, 2017-07-21:
Thanks so far. I would guess that the decision, wheather to start on a new byte or continue the last is based on the slave input/output size and not on the size of the channel right?

nakarlsson, 2017-07-21:
Yes, we have an open issue to force byte alignment.

closed 2017-07-21 by stefanpoeter