OpenEtherCATsociety/SOEM #120: EL6601 strange setup behavior: bug in EL6601 or missing feature in SOEM?

Apollo3zehn, 2017-07-23:
The last days I was trying desperately to configure the EL6601 ethernet switch port terminal and get it to OP mode. Besides normal Ethernet routing (via EoE), this terminal has also the capability to communicate via EAP protocol to the outer world and presenting the variables to the normal EtherCAT network. So basically it is similar to the EL6731-0010 Profibus slave (acting as a gateway) that I was already able to setup. I thought it would be easy to adapt the configuration but in fact it was very hard because I even did not get in running in TwinCAT.

Today I found a way to define a single subscriber (PDO 0x1A00) that is mapped to a single 2-byte input variable and its corresponding status variables. There are two additional predefined PDOs (0x1680 & 0x1A80) that describe the EAP state. So in total I have the following:

SM2 - mapping:
- 0x1680

SM3 - mapping:
- 0x1A00
- 0x1A80

And 0x1A00 is mapped to the following objects (they all belong to subscriber 1, variable 1):
0x6000:03 ("Sub-Var 1_Quality")
0x6000:04 ("Sub-Var 1_CycleIdx")
0x6000:05 ("Sub-Var 1_VarData")

To make this working I created the following SDO write request (in C#):
```
// Device Configuration Data (0xF800…0xF8FF)

    R_0xF800 = new object[] { (UInt16)0x02, (UInt16)0x0000, (UInt16)0x0100 };

// PDO configuration

    R_0x1680 = new object[] { (UInt16)0x0001, (byte)0x10, (byte)0x02, (byte)0x00, (byte)0xF1 }; // PDO control

    R_0x1A00 = new object[] { (UInt16)0x0003, // Var 1
                                (byte)0x10, (byte)0x03, (UInt16)0x6000,
                                (byte)0x10, (byte)0x04, (UInt16)0x6000,
                                (byte)0x10, (byte)0x05, (UInt16)0x6000 };

    R_0x1A80 = new object[] { (UInt16)0x0001, (byte)0x10, (byte)0x01, (byte)0x00, (byte)0xF1 }; // PDO status

// PDO Assign objects (0x1C12, 0x1C13)

    R_0x1C12 = new object[] { (UInt16)0x0001, (UInt16)0x1680 };
    R_0x1C13 = new object[] { (UInt16)0x0002, (UInt16)0x1A00, (UInt16)0x1A80 };

// Receiving Frame Data

    R_0x6000 = new object[] { (byte)0x05,
                                (byte)0x0A, (byte)0xFC, (byte)0x81, (byte)0x04, (byte)0x02, (byte)0x01, (byte)0x01, (UInt16)0x0001, (byte)0x00, (byte)0x00, (byte)0x00, (byte)0x00, (byte)0x00, (byte)0x00  };

// Receiving Frame Identification (Ignore Item Net Var Subscriber)

    R_0x6002 = new object[] { (UInt16)0x0005,
                            (byte)0x01, (byte)0x00, (byte)0x01, (byte)0x01, (byte)0x01 };

// Receiving Frame Length (Area Length Nat Var Subscriber)

    R_0x6003 = new object[] { (UInt16)0x0005,
                            (UInt16)0x0006, (UInt16)0x0002, (UInt16)0x0002, (UInt16)0x0002, (UInt16)0x0002 };

// Configuration Data object area (0x8000…0x8FFF)

    R_0x8000 = new object[] { (UInt16)0x03 };        
```

and then I applied the configuration via ecx_SDOwrite in a certain sequence:

```
// not important here
new SdoWriteRequest(0xF800, 0x00, R_0xF800),

// to create a subscriber (i.e. input) variable, creates the OD entry 0x6003
new SdoWriteRequest(0x8000, 0x04, R_0x8000), 

// defines the data type and other stuff, creates the OD entries 0x6000 and 0x6002
new SdoWriteRequest(0x6003, 0x00, R_0x6003),

// contains "quality", "cycleidx" and the received data "vardata" that were mentioned before
new SdoWriteRequest(0x6000, 0x00, R_0x6000),

// not important here
new SdoWriteRequest(0x6002, 0x00, R_0x6002),

// PDO configuration
new SdoWriteRequest(0x1A00, 0x00, R_0x1A00),
new SdoWriteRequest(0x1680, 0x00, R_0x1680),
new SdoWriteRequest(0x1A80, 0x00, R_0x1A80),

// Sync manager mapping
new SdoWriteRequest(0x1C12, 0x00, R_0x1C12),
new SdoWriteRequest(0x1C13, 0x00, R_0x1C13)
```

The strange things begin now. When inspecting with Wireshark, one can see that the registers 0x8000, 0x6003, 0x6000, 0x6002, 0x1A00, 0x1680 and 0x1A80 are **only writable**, altough mentioned differently in the documentation. A read attempt fails with SDO error 0x06020000 (The object does not exist in the object directory). The entries 0x1C12 and 0x1C13 are in fact readable but they return garbage data that was written in previous SDO calls.

So for me it looks like that all these entries are more like _commands_ or triggers that are not meant to be read. Normally this is no problem because I don't need to read the content of these registers. But SOEM needs to read these registers during **ec_config_map**. Since the terminal aborts the SDO upload, the **Ibits** and **Obits** variables are zero and therefore the sync manager length calculation is also zero, which disables the sync managers entirely.

Then I tried to make an offline configuration using the **ethercatconfiglist.h** file. But this leads to another unexpected behavior: When using the config list, SOEM does not set the CoE-details field. Therefore it does not know that the EL6601 has complete access (CA) ability. But the beforementioned objects can only be written via CA (or I did any mistake that prevented legacy access). Also this option is only feasable if I have only one EL6601.

So my only "solution" (which is not a solution) to get finally into OP mode, is to hard code the actual sync manager length in the function **ecx_map_coe_soe**. It looks then like this:

```
if (slave == 4)
   {
	   context->slavelist[slave].Obits = 16;
	   context->slavelist[slave].Ibits = 64;
	   context->slavelist[slave].SM[2].SMlength = 2;
	   context->slavelist[slave].SM[3].SMlength = 8;
   }
   else if (!context->slavelist[slave].configindex)
   {
      Isize = 0;
      Osize = 0;
      ...
   }
```

So either this is a bug in EL6601 (revision 0x00110000) or SOEM is not aware of this behavior. A real solution could look like that the following condition (that comes directly after the PO2SOconfig hook execution)

```
if (!context->slavelist[slave].configindex)
   {
     ... // execute ecx_readPDOmap and so on ....
   }
```

 is changed to: 

```
if (!context->slavelist[slave].configindex && (!Isize && !Osize))
   {
     ... // execute ecx_readPDOmap and so on ....
   }
```

Another option would be to move 

```
Isize = 0;
Osize = 0;
```
outside the "if" block, before the hook is executed.

What do you think?

nakarlsson, 2017-07-24:
It sounds like a buggy slave, something "hang/crash" the slave during your configuration 
Or 
you're missing some important commisioning information since your calls to 1c12 and 1c13 don't respond with proper data.

Did you manage to start with TwinCAT?
Have you asked Beckhoff support?

Apollo3zehn, 2017-07-24:
Yes all SDO calls succeed, but only if I do it in a certain order since some entries are created by previous calls. 

Today I updated my TwinCAT version, now it works completely with TwinCAT. Another issue (only with SOEM) is that I can get into OP only once, the second time fails. I will do some more testing with TwinCAT to compare the differences with SOEM.

I talked to Beckhoff support today because of some other questions I had to that terminal. Probably I will get a new firmware version the next days to enable other features. Maybe this solves at the sane time the writeonly problem.

nakarlsson, 2017-07-24:
You should be able to check the required pre-configuration in TwinCAT, try to apply those steps from SOEM. Please keep us updated and provide a pull request if needed.

Apollo3zehn, 2017-07-24:
Yes I used the TwinCAT startup list but that list was wrong until I updated TwinCAT today. 

But Wireshark will definitely help. I will keep you updated.

Apollo3zehn, 2017-07-26:
I got the new firmware and did more testing and analyzing the last days. The firmware update (now: software version 07, hardware version 09, revision 18) did not change anything related to that issue. 

The other problem that I could not restart the system two times in a row was caused by a faulty design in my application. I was not using the PO2SOconfig hook. Instead I had my own logic between **ecx_config_init** and **ecx_map_coe_soe**. The last thing **ecx_config_init** does is to request PreOp and the first thing **ecx_map_coe_soe** does is to wait for actually reaching PreOp. So I was in the middle of that. And it worked until now. But as this slave is very slow, the bug appeared now in my application. 

The original problem changed a little bit. 1C12 and 1C13 respond now with the correct PDO addresses (I ensure now that PreOp is reached before talking to mailbox). But SOEM goes further and tries to read the PDO content (e.g. 0x1A00), and this is what still fails. I will redesign my application and ask in the ETG forums if this is the default behavior.

nakarlsson, 2017-07-26:
If you repeat the twincat startup list from your config hook, doesn't that work? I bet reading1C12,1C13 fails cause you are required to configure them properly, in the end TwinCAT must do the same. 

Can you post the P->S config either from TwinCAT or the ESI, should be the same.

Apollo3zehn, 2017-07-27:
I am repeating the TC startup list exactly. The difference between SOEM and TwinCAT is that TwinCAT never reads the PDO content. I guess it knows the input and output size from the startup list.

Today I discovered that I can easily access e.g. PDO 0x1A00 as soon as the slave is in SafeOp. I tried the following:

- Put slave into **OP** with TwinCAT. Downloading of PDO 0x1A00 with SOEM **succeeds**. TwinCAT shows the PDO content in "CoE-Online" tab.
- Put slave into **SafeOp** with TwinCAT. Downloading of PDO 0x1A00 with SOEM **succeeds**. TwinCAT shows the PDO content in "CoE-Online" tab.
- Put slave into **PreOp** with TwinCAT. Downloading of PDO 0x1A00 with SOEM **fails**. TwinCAT does not anymore show the PDO content in "CoE-Online" tab.

These results are support by the Wireshark image below: I modified SOEM to read the PDO content right before requesting SafeOp and right after reaching SafeOp. In-between, the AL status is read two times. One cannot see it in the screenshot, but the first 0x130 read returns "PreOp" and the second one returns "SafeOp". And right after that, the PDO 0x1A00 can be read. 

[image]

___________________________________________________________________________________

The next picture shows the startup list of TwinCAT:

[image]

And here are two Wireshark dumps, 1x with SOEM, 1x with TwinCAT. There you can see that TwinCAT never tries to read the PDO content. 

[https://my.syncplicity.com/share/wtaccumn0tb62kw/SOEM](url)
[https://my.syncplicity.com/share/izb6xjga6k71mbo/TC](url)
Password: 123456

nakarlsson, 2017-07-27:
TwinCAT know the length from reading the ESI I think, and by not allowing SOEM to do that online is a faulty behaviour, in my opinion.

nakarlsson, 2017-08-17:
Any update? I saw your post on ETG forum, did you mail Beckhoff aswell?

nakarlsson, 2017-09-30:
Ok, to close?

Apollo3zehn, 2017-09-30:
Sorry I am currently running out of time. I will progress on this issue later so you can close it for now. If it turns out to be related to SOEM I will create a new one.

closed 2017-09-30 by nakarlsson