OpenEtherCATsociety/SOEM #121: question about config map

joeization, 2017-07-27:
After i managed to receive/send pdo, i notice that the iomap size is weird.

If i connect slave with soem first, it will get a smaller map size.
But if i use the manufacturer's tool to connect first then disconnect, soem then can get correct map size.

Is there any function that i can force set the map size?

i've try to set input/outputsize before iomap, but its not work.

nakarlsson, 2017-07-27:
You probably need to do some commisioning of the slave while in PREOP before entering SAFEOP.

Her is an example used to configure an Moog drive with the SOEM P2S hooks. https://github.com/hefloryd/SOEM/commit/22d8f5d

joeization, 2017-08-02:
i figure out how to set it, thanks.

closed 2017-08-02 by joeization