OpenEtherCATsociety/SOEM #127: Configuring Ax5206 drive to safe op mode

mahantheshmvh, 2017-09-01:
Hi,
We are facing problem to configure AX5206 to Safe-op mode and Op mode using SOEM.
We are getting Drive error as 0xF4A5: A parameter error has been detected in the SoE communication layer".
We are loading the Mailbox Init commands present in config file exported  from TwinCAT using ec_SoEWrite.

I have sent Wireshark dumps to openethercatsociety@rt-labs.com
Can anyone help us with this issue.?

mahantheshmvh, 2017-09-01:
Slave info As shown below:
tarting slaveinfo
ec_init on enp2s0 succeeded.
1 slaves found and configured.
Input Bytes: 20 Output Bytes : 12  
Calculated workcounter : 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=   1 : Unspecified error

Slave:1
 Name:AX5206-0000-0200
 Output size: 96 bits
 Input size: 160 bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
Configured address: 1001
 Man: 00000002 ID: 14566012 Rev: 00c80000
 SM0 A:1800 L: 128 F:00010026 Type:1
 SM1 A:1a00 L: 128 F:00010022 Type:2
 SM2 A:1000 L:  12 F:00010024 Type:3
 SM3 A:1100 L:  20 F:00010022 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000c Ll:  20 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 18
 CoE details: 00 FoE details: 01 EoE details: 00 SoE details: 02
 Ebus current: -250[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM2 RXPDO 0x0018 MDT 1
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x0086:0x00 0x10 UNSIGNED16   Master control word
  [0x0002.0] 0x002F:0x00 0x20 INTEGER32    Position command value
  SM2 RXPDO 0x1018 MDT 2
     addr b   index: sub bitl data_type    name
  [0x0006.0] 0x0086:0x00 0x10 UNSIGNED16   Master control wor
  [0x0008.0] 0x002F:0x00 0x20 INTEGER32    Position command value
  SM3 TXPDO 0x0010 AT 1
     addr b   index: sub bitl data_type    name
  [0x000C.0] 0x0087:0x00 0x10 UNSIGNED16   Drive status word
  [0x000E.0] 0x0033:0x00 0x20 INTEGER32    Position feedback 1 value
  [0x0012.0] 0x00BD:0x00 0x20 INTEGER32    Following distance
  SM3 TXPDO 0x1010 AT 2
     addr b   index: sub bitl data_type    name
  [0x0016.0] 0x0087:0x00 0x10 UNSIGNED16   Drive status word
  [0x0018.0] 0x0033:0x00 0x20 INTEGER32    Position feedback 1 value
  [0x001C.0] 0x00BD:0x00 0x20 INTEGER32    Following distance
End slaveinfo, close socket

chimeno86, 2017-09-22:
hello, I also have problem with the AX5000, I have AX5203 and AX5206, I managed to work one part but another not, can I help? have you solved it?

mahantheshmvh, 2017-09-23:
Yes please.
Am  getting Drive error as 0xF4A5: A parameter error has been detected in the SoE communication layer".
I exported the Mailbox Init commands from twincat. The same is sent to drive using ec_SoeWrite function.

chimeno86, 2017-09-23:
hello, the truth is that these AX5000 drivers are a bit complicated to handle by the way they are made, one of the things you can try is to test, is to load only the essential IDN configuration and add parameter per parameter which is the one that you error, also if you want you can pass me your code and I prove it in my AX5206 in case it is a code problem.
regards

mahantheshmvh, 2017-09-23:
> Bug from
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png","action":{"name":"Open
> GitHub","url":"https://github.com/OpenEtherCATsociety/SOEM"}},"updates":{"snippets":[{"icon":"PERSON","message":"@chimeno86
> in #127: hello, the truth is that these AX5000 drivers are a bit
> case it is a code problem.\r\nregards"}],"action":{"name":"View
> Issue","url":"https://github.com/OpenEtherCATsociety/SOEM/issues/127#issuecomment-331620482"}}}
> This message has been scanned for viruses and
> dangerous content by *MailScanner* <http://www.mailscanner.info/>, and is
> believed to be clean.

Hi,

The only essential IDNS required to bring AX5206 to Op mode are:

      uint8 SoEWrData4[8] ={0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00 };

    retval += ec_SoEwrite( 1, 0, 64, 32778, 8, SoEWrData4, EC_TIMEOUTRXM 
);// 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00

     retval += ec_SoEwrite ( 1, 1, 64, 32778, 8, SoEWrData4, 
EC_TIMEOUTRXM );//  0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00

     uint16 SoEWrData5 = 0x07D0;
     retval += ec_SoEwrite ( 1, 0, 64, 1, 2, &SoEWrData5, EC_TIMEOUTRXM 
);// D0 07

     SoEWrData5 = 0x07D0;
     retval += ec_SoEwrite ( 1, 0, 64, 2, 2, &SoEWrData5, EC_TIMEOUTRXM 
);// D0 07

     uint16 SoEWrData6 = 0x000B;
     retval += ec_SoEwrite ( 1, 0, 64, 32, 2, &SoEWrData6, EC_TIMEOUTRXM 
);// 0B 00

     retval += ec_SoEwrite ( 1, 1, 64, 32, 2, &SoEWrData6, EC_TIMEOUTRXM 
); //0B 00

I have attached code in which you can through InitEtherCAT function used 
for bringing slaves Slaves to op mode.

If you have working code please share with me.

I tried sending each parameter one by one but the error remains the same.

-- 
Best wishes,
Mahanthesh
R&D Engineer | Pragati Automation

chimeno86, 2017-09-24:
Hello again,
my email chime69@msn.com, you send me the code to prove it, if you can also pass me your configuration of the AX5206, in TWINCAT3 in section DRIVE1 (AX5206-0000-0200) STARTUP tab you export everything and I pass it to see your configuration, it is also important the parameters of SYNC0 and SINC1, if they are not good also not to have if I can help you? What project are you doing to use these ServoDrivers?...

chimeno86, 2017-09-25:
the basic configuration that you must send to OP:
S-0-0015 axis 0 axis 1
S-0-0016 axis 0 axis 1
S-0-0024 axis 0 axis 1
S-0-0001 axis 0
S-0-0002 axis 0
S-0-0032 axis 0
P-0-0304 axis 0
P-0-0201 axis 0
P-0-0202 axis 0
P-0-0203 axis 0
P-0-0204 axis 0

Sync0 = 250,000
Sync0_Shift = 0
Sync1 = 1,750,000
Sync1_shift = 0

regards

mahantheshmvh, 2017-09-26:
Hi,
I have sent the start up list exported from TwinCAT to your mail.
I have C++ code to communicate with Drive . We are reading the config XML export (from TwinCAT ) in our code.
We are working with U++ IDE with Xenomai  and rtnet.
We have same Sync data.
Sync0 = 250,000
Sync0_Shift = 0
Sync1 = 1,750,000
Sync1_shift = 0

Shall i send our C++ code ?

chimeno86, 2017-09-27:
hello, I do not work in that code, in the code I can not help you much, you have configured it only with the previous list? go OP?

mahantheshmvh, 2017-09-29:
Hi,
Both the lists are same.  The idns in the start up list are saved as hex values in the xml file exported from twincat. 
I sent all the idns as in start up tab to the drive but could not  reach 
Safe op. 
Can you share your drive initialisation code ?

chimeno86, 2017-10-03:
Hello, I have not used SOME for some time, I did not answer the question, I could not get it, I moved to another system ...

chimeno86, 2017-10-19:
hello again, I think you would have to increase the waiting time to op because it needs more time.

nakarlsson, 2017-11-11:
Seems like its not open anymore

closed 2017-11-11 by nakarlsson

jon4729, 2019-04-30:
hello,
I have a similar issue with the configuration of the AX5206 form SafeOP to OP with SOEM. Could you please share your code for the configuration?

mahantheshmvh, 2019-05-02:
> GitHub","url":"https://github.com/OpenEtherCATsociety/SOEM"}},"updates":{"snippets":[{"icon":"PERSON","message":"@jon4729
> in #127: hello,\r\nI have a similar issue with the configuration of
> code for the configuration?"}],"action":{"name":"View
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url":

Afetr many attempts also i failed in configuring from SafeOp to Op . Now 
i stopped working on it.

I will share the code file.

-- 
Best wishes,
Mahanthesh
R&D Engineer | Pragati Automation

jon4729, 2019-05-02:
hello,
thank you for your response. Could you please send the file to dms@email-wizard.com ?