OpenEtherCATsociety/SOEM #130: Some basic questions about SOEM

Norber96, 2017-09-25:
Good night.
I am a student trying to learn how to use SOEM to control a PLUTO servodrive.
There are some basic things that i am not really understanding so i hope you can help me:
1- Currently i am able to read SDO data from the servodrive like the max temperature, etc. But I don't understand how to read PDO data, i know that EtherCAT creates a map and that i can access it with the function ec_send_processdata( ), but i don't get how i can for example read the actual temperature that my servodrive has. 
2-I have the object dictionary that ingenia provides but my servodrive doesn't support SDO info so i dont have and object dictionary loaded in the servodrive. Is there a way to load the .xml in the servodrive to make things easier?

Sorry if the questions are too basic, but i really need to get this to work i have read the tutorial and a lot of projects from other people but i don't really understand this concepts.

Thanks for your time.

nakarlsson, 2017-09-30:
You can program SDOs manually as described in this example. https://github.com/hefloryd/SOEM/commit/22d8f5d

Norber96, 2017-09-30:
Hey thanks for replying.

I do understand how to program SDOs, but thanks for the example it is very ilustrative.
My problem is that I don't really get how to read/write PDO and how to update them. I have read some examples but i don't get the steps. I have seen that you create a buffer for inputs and for outputs. Then you update that buffer with functions send_processdata(), and receive_processdata(). But i don't know how to assign the PDO into that buffer and how to read or write them afterwards. Is the read/write done the same way as the SDO data, with the functions SDO_read and SDO_write?, or is there another easier way to deal with the PDO?.
If you could show me the steps i have to follow to make this work it would be awesome.

Sorry for the basic concepts and thanks for your time.

nakarlsson, 2017-09-30:
Check test/linux/red_test.c for example, there you can se how to connect to the IOmap through the ec_slave struct.

/* check for EL2004 or EL2008 */

SOEM rely on slaves having online PDO info, if not you need to use the configlist, ec_configlist.h to program the SM and FMMU according to the XML.

For slaves with OD you can use test/linux/slaveinfo -map to list the PDO layout.

Norber96, 2017-10-01:
Hey  about the online PDO info, I have read the redtest, and i suppose here is where it uploads the info of the PDO:
" if (dorun>0)
      {
         wkc = ec_receive_processdata(EC_TIMEOUTRET);

         dorun++;
         /* if we have some digital output, cycle */
         if( digout ) *digout = (uint8) ((dorun / 16) & 0xff);
"

Is this enough to update the information of the PDO? Then how do i read the updated information, with the SDO_read function or is there a specific function?.

Thanks for your time

nakarlsson, 2017-10-01:
Yes, you read / write PDO info inbetween calls to send/receiveprocessdata. This functions exchange the IOmap eg PDOs

Norber96, 2017-10-01:
So, i can read for example the actual position with the function SDO_read, while doing the send/receiveprocessdata, and i will get the real position, cant i?. I will try to implement it in my program and see if it works.

Thanks for your time.

nakarlsson, 2017-10-02:
No, don't use SDOread, the actual position is part of the IOmap. Use the ec_slave[i].inputs as reference to your inputs PDOs, how to interpret the PDOs is up to your slave, think of it as a packed struct holding position and more? Is for you to figure out with help from XML or Object Dictionary.

nakarlsson, 2017-10-07:
Can we close the issue?

Norber96, 2017-10-07:
Hey sorry for not replying.
I have been trying to do what you told me but i am unable to write in PDOs using the IOmap. Can you show me a example of how, for example you read/write using the IOmap?.
I have read the tutorial and i suppose, this is where it writes in the PDO, but i don't understand the code:
```
#defineEL4001_1           2
...
void set_output_int16 (uint16 slave_no, uint8 module_index, int16 value)
{
   uint8 *data_ptr;

   data_ptr = ec_slave[slave_no].outputs;
   /* Move pointer to correct module index*/
   data_ptr += module_index * 2;
   /* Read value byte by byte since all targets can't handle misaligned
 addresses
    */
   *data_ptr++ = (value >> 0) & 0xFF;
   *data_ptr++ = (value >> 8) & 0xFF;
}
...
set_output_int16(EL4001_1,0,slave_EL4001_1.out1)
```; 
Could you explain me the code, please?

Sorry for the basics but I'm a student and it is difficult for me to understand all this concepts.

Thanks for your time.

nakarlsson, 2017-10-07:
Her is another simple example

Declare the slave PDO, example here is a 2 channel Analog Output

The slave PDO map
typedef struct PACKED
{
	int16 output1;
	int16 output2;
} out_EL4002;

out_EL4002 *aout16;

Find the slave and set the variable
...
 /* Locate the slave */

    for (slc = 1; slc <= ec_slavecount; slc++)
    {
           if ((ec_slave[slc].eep_man == 0x00000002) && (ec_slave[slc].eep_id == 0x0fa23052))
           {
                  printf("Found %s at position %d\n", ec_slave[slc].name, slc);
                  aout16 = (out_EL4002*)ec_slave[slc].outputs;
                  ao_slave = slc;
            }
...

Set the values in the IOmap
...
      aout16->output1 = 0x7FFF * (int16)(ao1_volt * 10) / 100;
      aout16->output2 = 0x7FFF * (int16)(ao2_volt * 10) / 100;
...

Norber96, 2017-10-10:
Hey thanks for answering, i tried it and it worked, I'm really grateful.

Although, i have one question about the code:
`aout16->output1 = 0x7FFF * (int16)(ao1_volt * 10) / 100;`
Is this a required structure? I mean why do u multiply by 0x7FFF? 

And one last thing, i have trying to change the value of the controlword to be able to start the control, but in this object i need to change the bits instead of the value in hexadecimal, is it possible that i insert the structure of bits that i want in binary or do i have to change it to hexadecimal?.

Thank you so much for your time.

nakarlsson, 2017-10-10:
Those questions have nothing to do with EtherCAT, you're close and as a student you need to figure out somethings yourself :)

closed 2017-10-10 by nakarlsson