OpenEtherCATsociety/SOEM #131: SOEM configuration, compilation and debugging issues in INtime.

GREduardo, 2017-09-29:
I am trying to use SOEM for INtime and have some questions in this regard. 

I do not see code in the CMakeList.txt referring to the INtime compilation, so, for the moment, I created an INtime project on Visual Studio 2017 adding SOEM C files, header directories and library references. Several compilation errors appeared after this (like Error       C2061    syntax error: identifier 'ecx_portt'). I made most of this errors disappear by adding 

#include "nicdrv.h"

at the top of ethercatmain.h. 
 >>Is this a clean solution?

After this I had errors like:
E0020    identifier "osal_mutex_t" is undefined

Since functions using osal_mutex_t are not used anywhere I just commented this section. 
>>Again, is this a clean solution?

The only error left was for a definition of the function "osal_thread_create" and was solved with a new definition of this function.

Nevertheless, three cases happened after the successful compilation with a minimal code (ecx_init(&ctx[0], "rtl1g0"), ecx_config_init(&ctx[0], FALSE)). 

1. Code runs successfully.
2. ecx_config_init(&ctx[0], FALSE) returns <0 (every time after the first time this appears).
3.Exception 19 (Thread is using all processor time) is shown next to some particular functions and then also 2 happens.

>>Have you ideas for the solution of this issue?
>>Can you please tell me tell me how can I use CMakeList.txt to create a SOEM project for INtime?

I will appreciate any help you can provide me

nakarlsson, 2017-09-29:
The source is for InTime 4, we also used an InTime project and you can arrange the code as you like.

2 make sure you release/close you NIC in between
3 you need some sleep/yield for another thread between send/recv othwrwise you use 100% for sure.

GREduardo, 2017-10-05:
thank you nakarlsson
I have some updates:

first for:
3.Exception 19 (Thread is using all processor time) is shown next to some particular functions and then also 2 happens.

adding a sleep function inside any of the functions where problem is pointed make this message dissapeare. 

Now for 

2.ecx_config_init(&ctx[0], FALSE) returns <0 (every time after the first time this appears).

This problem seems to be related with an overruned resource. Adding more sleep functions inside ecx_config_init seems to help, but I think it is not the solution. 
The above helped me to put the slave into OPERATIONAL state more often allowing me to test the next code:

for (i = 1; i <= 1000; i++) 
{
       *(ec_slave[1].outputs) = 0xFF;
       ecx_send_processdata(&ctx[0]);
       knRtSleep(10);
       ecx_statecheck(&ctx[0], 0, EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE);
       printf("EtherCAT slave %x %x \n", ctx[0].slavelist[0].state, ctx[0].slavelist[1].state);
}
 
Which after some iterations returns:
…
EtherCAT slave 8 12
EtherCAT slave 8 12
EtherCAT slave 0 12
EtherCAT slave 0 12
EtherCAT slave 0 12
…

Outputs signals are successfully received by the slave until the state change to 0.

I went deeper into the code and I realized that function 

hpeGetReceiveBuffer(port->handle, &rxbuffer);

returns an error (between the change of state)and that 

rxbuffer->used

changes from the value 0x3c to 0x457dd4 and gets a higher value in further iterations.

nakarlsson, 2017-10-16:
Ok to close? Seems like a driver issue or driver implementation that needs to be update to fit your InTime version

GREduardo, 2017-10-17:
It is ok to close. I changed ec_init() function to set a different configuration port and etherCAT communication started to work.

closed 2017-10-17 by GREduardo