OpenEtherCATsociety/SOEM #133: Multi slave communication problem

TXueTHU, 2017-10-16:
Hi,
    A copley driver and an arducat board were connected to one ethercat network.
    I followed the simple_test example and some other references to config slave pdo mapping etc and the driver works well. However, the data from arducat board didn't update. 
   I tried to get the ethercat network linked to ethercat configurator (Twincat). When the driver works under DC Cyclic , enable SYNC1 and arducat board is in DC synchron mode , the slaves works fine. Also when driver is in  free run mode  and arducat is under smx synchron mode  , the two slaves are happy. Otherwise, the driver works fine while the arducat didn't update the data.
   I'm wondering how to config the slave Operation Mode (Asyn Mode) with soem? What should i do to make the arducat board work?
    Look forward to your generous repley!

nakarlsson, 2017-10-16:
You need to consult the manual on howto comission you slave and here is an old issue explaining howto configure slaves.

TXueTHU, 2017-10-17:
Thanks for your fast repley. 
I reviewed the old issues but didn't find the similar problem. 
My problem is how to config the slave sync mode to trigger the slaves updating and sending processing data to master. 
I followed the simple_test code and use slave[1].inputs,slave[2].inputs to read the process data. The first slave works well while the second didn't update the data.

nakarlsson, 2017-10-17:
There are a lot of "issues" explaining howto configure slaves, you need to consult your slave reference manual commissioning chapter on how you should configure your slave to behave as expected.

You say you need the slave to run in Asyn Mode, the ref manual is the place to look for how to do that, if possible.

TXueTHU, 2017-10-17:
OK, Got it. Will try that. 
Thanks a lot for your help.

nakarlsson, 2017-10-17:
Here is an example for configure a slave.

https://github.com/OpenEtherCATsociety/SOEM/issues/120

nakarlsson, 2017-11-02:
Nothing new

closed 2017-11-02 by nakarlsson

TXueTHU, 2017-11-04:
I fixed the above problem and it's the hardware problem.
I write ec_dcsync0() after ec_configdc()  to the copley driver slave , sometimes it works and sometimes not. Do you know what's wrong with it ?

nakarlsson, 2017-11-04:
What is the AL StatusCode?

TXueTHU, 2017-11-05:
AL StatusCode means what?