OpenEtherCATsociety/SOEM #134: Problem in the state machine

MrArdilla, 2017-11-02:
Good morning.

I own a Pluto servodrive, the current problem I have is that I can't change between the states "Ready to switch on" and "Switch on enable", whenever I send the controlword 7 or F the servodrive returns to the state "Switch On disabled".

I have read the documentation of my controler and it doesn't have any useful information about this error, I have tried with Twincat but when i try to online write the controlword it doesn't change, I mean i write 7 and it stays at 6, and the same happens when i try to wirte it with SDO instead of PDO.

Any help is help is welcome.

Thanks for your time.

nakarlsson, 2017-11-03:
What does your PDO mappings look like, try running slaveinfo -map

You should do the following according to the ESI, have you done that? I can't find out exactly what it does but it might be mandatory stuff.
            <InitCmd>
              <Transition>PS</Transition>
              <Index>#x60C2</Index>
              <SubIndex>1</SubIndex>
              <Data>01</Data>
              <Comment>Interpolation time period</Comment>
            </InitCmd>
            <InitCmd>
              <Transition>PS</Transition>
              <Index>#x60C2</Index>
              <SubIndex>2</SubIndex>
              <Data>FD</Data>
              <Comment>Interpolation time index</Comment>
            </InitCmd>

MrArdilla, 2017-11-06:
Hey, thanks for replying.

I have attached the map, if you could take a look at it. But i think it is okay, besides the normal parameters i have added the DC link circuit voltage to monitor. Because sometimes Twincat has given me an error related to voltage, but it stays stable and in the interval it is supposed to be.
*Don't mind the spaces, it is something i used to check the wkc.*
[image]

Besides, i have checked the parameters you told me with the MotionLab and SOEM, and the values i got are in line with the manual, there's a photo with the values as well.

[image]

What do you mean with doing the following according to the ESI? Ingenia has given me a .xml file and the parameters i get match the control position and the paramater i have added, but I think I haven't done a configuration related to the ESI,

Any is welcome, thank you so much.

nakarlsson, 2017-11-06:
Can you reach OPERATIONAL? 
If then I think you got some non EtherCAT related issue
If not what is the ALStatusCode.

MrArdilla, 2017-11-07:
Yes, I can reach OPERATIONAL with no problem, i run the simple_test and it works perfectly.
Okay thanks for your time, i will take a look at it.

Thank you so much.

nakarlsson, 2017-11-11:
No SOEM or EtherCAT issue, maybe commisioning issue

closed 2017-11-11 by nakarlsson