OpenEtherCATsociety/SOEM #135: Windows 10 issue

Cifi79, 2017-11-02:
I used SOEM with Win7 and some Beckhoff slave. Now I install the same application inside a new pc with Win10 and I have some problem.
I found that the problem was a filter called "Intel Technology Access Filter Driver", when I disabled this option inside the network board properties, ethercat starts to work.

Does anyone known what is this option?
Do you have any suggestion reguarding the usage of SOEM?

nakarlsson, 2017-11-03:
What have you found out? I've never come across that settings before and I've an Intel NIC in my laptop running SOEM via Winpcap.

Cifi79, 2017-11-08:
I saw that the led of network activity in my slave didn't start to blinking with that Intel filter active.
But in my pc I have a Realtek NIC because I know that it's not important wich chipset have the network board. I heard that during an ethercat roadshow, I asked directly to mr. Martin Rostan,
Executive Director ETG.

Another question, can you use the DC method using a non-realtime windows OS (i.e. win10)?

nakarlsson, 2017-11-08:
Can't say, depends on the DC application.

nakarlsson, 2017-11-11:
No more questions

closed 2017-11-11 by nakarlsson