OpenEtherCATsociety/SOEM #137: soem slave status checking problem with ros kinetic and ubuntu 16.04

junhyeokahn, 2017-11-24:
Hi,
I am currently using ros kinetic with ubuntu 16.04.
Thanks to your open source policy, I could download and run your example code slaveinfo.c.
When I testing with 7 slaves, it works well.
However, when I copy the code and generate a ros package, it behaves differently.
The different part happens in the following part of the code.
```
ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 3);
if (ec_slave[0].state != EC_STATE_SAFE_OP )                                                                                                                                                        
{
      printf("Not all slaves reached safe operational state.\n");                                                                                                                                     
      ec_readstate();                                                                                                                                                                                 
      for(i = 1; i<=ec_slavecount ; i++)                                                                                                                                                              
      {                                                                                                                                                                                               
           if(ec_slave[i].state != EC_STATE_SAFE_OP)                                                                                                                                                    
           {                                                                                                                                                                                            
               printf("Slave %d State=%2x StatusCode=%4x : %s\n",                                                                                                                                        
                         i, ec_slave[i].state, ec_slave[i].ALstatuscode, 
                         ec_ALstatuscode2string(ec_slave[i].ALstatuscode));                                                                                     
            }                                                                                                                                                                                            
      }                                                                                                                                                                                               
}
```
When I run the code with ros, ec_slave[0].state is not equals to EC_STATE_SAFE_OP, so it goes into if statement. One more thing I notice is that when I use small number of slaves (e.g. 3 instead of 7), ec_slave[0].state equals to EC_STATE_SAFE_OP.
 
However, without using ros package, ec_slave[0].state is always equals to EC_STATE_SAFE_OP.
Is there any problem when I try to bind soem to ros package?

nakarlsson, 2017-11-24:
Sounds like a timing/performance issue.

Make sure you have EC_MAX_MAPT 1
Take a wireshark log in both cases, can give a hint.

junhyeokahn, 2017-11-24:
Thank you.
It absolutely solves the problem.

closed 2017-11-24 by junhyeokahn

nakarlsson, 2017-11-25:
Using more threads during configuration require some timeout relaxation. Up to the user to tune in if needed.
Thats Why 1 thread is recommended by default.