OpenEtherCATsociety/SOEM #138: Slave doesn't switch to OPERATIONAL STATE with no error code

mikekaram, 2017-12-11:
Hi,
I'm newbie in EtherCAT so please be gentle:). I'm trying to control an XMC4800 from a PC via EtherCAT. I'm running the simple_test and I'm having this kind of output:
[image]
while the dump from Wireshark is this : 
[simple_test_no_op.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1548818/simple_test_no_op.pcapng.txt)
(note: I have changed chk to 100 in an attempt to solve the issue)
The slaveinfo -map gives me this output:
[slaveinfo_no_op.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1548803/slaveinfo_no_op.txt)

and the eepromtool -i this output:
[image]

I don't know where (and what)  to look for this issue. 
Please help,
Mike

nakarlsson, 2017-12-12:
Your EtherCAT slave stack don’t work properly. You can check the SOES project as an alternative to the SSC stack.

mikekaram, 2017-12-12:
But the thing is that it worked with TwinCAT... What did you find and you are so sure?
Thank you very much for your reply...
Mike

nakarlsson, 2017-12-12:
I haven’t done a deep analysis, but you say it refuse to go to OP without an error, that is not a correct behaviour. 

Have you run conformance test tool to make sure it is properly implemented?

mikekaram, 2017-12-12:
No, I haven't...

mikekaram, 2017-12-12:
Anyway, I will test it and I will respond back...
Thank you very much for your help @nakarlsson ,
Mike

nakarlsson, 2017-12-12:
I checked the wireshark log, SOEM request OP and your slave stay in SAFEOP without ERROR. Either you miss events or your error reporting logic is faulty.

I’ve heard ”But it works with TwinCAT” so many times... Doesn’t mean it is a correct implementation.

mikekaram, 2017-12-12:
Ok thank you very much @nakarlsson ,
I will run conformance test tool to check and I will respond back,
Mike

nakarlsson, 2017-12-12:
Not a SOEM issue, the slave don't act on the OP state change, either by changing state or reporting error.

Contact your SSC provider.  We can help you with general slave questions and an alternative solution with SOES.

closed 2017-12-12 by nakarlsson

Rainer-Hoffmann, 2017-12-19:
Based on the wireshark file the master configures the FMMU and SM in opposite direction.
FMMU0 is configured in write direction (frame 258) to address 0x1400 and the SyncManager(3) protecting 0x1400 is configured in read direction (frame 256)
FMMU1 is configured in read direction (frame 260) at address 0x1100 and the SyncManager(2) protecting 0x1100 is configured in write direction (frame 254)
So that can't work properly, which is definitely a master failure. 
The Cyclic frames also indicating that the basic process data exchange is not running, which is extremely unlikely even if the slave would not operate correct.

So my assumtion is that the slave is waiting for process data communication an therefore do not react on the state request to OP. 
Furthermore the master stops polling the state after 5sec but the default SafeOP to OP timeout is 10sec (which has to be used in that case because the SOEM seems to use only SII information).
By default the internal SafeOP to Op timeout in the SSC is 9sec so I'm pretty sure that you would get an Error SafeOP reaction with the State Code 0x1B (PD SM watchdog) after 9sec.

nakarlsson, 2017-12-19:
SOEM configure the slave based on the information provided by the slave, so I wouldn’t say it is a SOEM failure if the slave isn’t correct implemented.
Rainer, are the slave EEPROM and OD correct? If they’re then we can assume there is a bug in SOEM.

mikekaram, 2017-12-20:
Hello again @nakarlsson and @Rainer-Hoffmann ,

I finally made the simple demo APP of Infineon for the XMC4800 to work with SOEM. What I did wrong, was to use the  v5.12 instead of the v5.11 of SSC Tool . When I switched versions SOEM just worked fine. Although I'm curious why this is happening. Maybe you guys can answer that, obviously not me.

Thanks again both of you for the support,
Mike

Rainer-Hoffmann, 2017-12-20:
@nakarlsson: I didn't analyse the read EEPROM data. But the SM configuration is correct (based on the wireshark) and the FMMU configuration is not part of the EEPROM (only the FMMU types are defined). 
Furthermore the timeout is not part of the EEPROM so the SOEM has to use the default values (which is currently not the case).

@mikekaram : thanks for testing again. I don't think that it is a bug in the stack, I would compare the generated eeprom data. Feel free to send me the gernerated files so I can have a look.

mikekaram, 2017-12-20:
@Rainer-Hoffmann : I don't know where I should send you the files.

nakarlsson, 2017-12-20:
I think you guys can solve this without my help, but out of curiosity please send a copy to our openethercatsociety mail.

mikekaram, 2017-12-20:
[POTENTIOMETER_SLAVE_1.zip](https://github.com/OpenEtherCATsociety/SOEM/files/1574989/POTENTIOMETER_SLAVE_1.zip)

nakarlsson, 2017-12-20:
I think the problem is the EEPROM, see below.  However, when I check provided code it don't match the settings seen in this slaveinfo output, slaveinfo_no_op.txt. So somewhere along the way the data get incorrect. If you retry, erase the emulated eeprom and load again, I see no reason why it shouldn't work. Unless there is a bug in the XMC code.

From slaveinfo_no_op.txt
 SM2 A:1100 L:   9 F:00010064 Type:4 <- SM type inputs but with controlflags for an outputs SM
 SM3 A:1400 L:   9 F:00010020 Type:3 <- SM type outputs but with controlflags for an input SM

Rainer-Hoffmann, 2017-12-21:
@mikekaram : thanks for the files, I mendtioned the email you've addressed before (ethercat_ssc@beckhoff.com)

@nakarlsson : The eeprom is correct if you would have checked the wireshark scan and take a look at the read eeprom data you would see that the SyncManager settings are consistent. 

While checking the wireshark scan I've seen that the master also reads the object 0x1C00 (SyncManager Types). I don't know why the master is doing this because all informations are also available in the EEPROM. However the object data is inconsisten with the EEPROM data. My assumption is that the SOEM mixup the information to create the configuration data for the SyncManagers/FMMUs.

@nakarlsson : have you fixed the timeout values? And could you please check why the SOEM reads the object 0x1C00?

@mikekaram : I will send you an updated source file which contains the aligned object data.

nakarlsson, 2017-12-21:
@Rainer-Hoffmann, SOEM use the OD (incl 0x1C00) to create the PDO mappings if supported and SOEM favors CoE over SII. I think your assumption is correct. 
Why does the slave data get inconsistent? Isn't the OD, EEPROM and ESI generated from the same source?

We'll look into the timeouts, https://github.com/OpenEtherCATsociety/SOEM/issues/141

Rainer-Hoffmann, 2018-01-08:
In general it is the case that the OD, EEPROM and ESI are generated from the same source. but not in case of the object 0x1C00.