OpenEtherCATsociety/SOEM #139: ethercatconfig.c -> context->slavelist[slave].parent field understanding

Apollo3zehn, 2017-12-13:
I am a little bit confused how the parent field `context->slavelist[slave].parent` is supposed to work. If I have the following structure: 

Master -> [AnybusXGateway (2 Ports)](https://www.anybus.com/de/produkte/gateway-index/anybus-xgateway/detail/anybus-x-gateway-ethercat-slave---modbus-tcp-server) -> EK1100 -> EL1008 -> EL2008

What would be the desired output? I would expect:

- Master (0)
  - context->slavelist[0].parent = 0
- AnybusXGateway (1)
  - context->slavelist[1].parent = 0
- EK1100 (2)
  - context->slavelist[2].parent = 0
- EL1008 (3)
  - context->slavelist3].parent = 2
- EL2008 (4)
  - context->slavelist[4].parent = 2

But I am getting different results. How is `parent` defined? I would say a parent is every slave with more than 2 ports but I fear I misunderstood this.

nakarlsson, 2017-12-14:
If you get different results for the same network the HW must not work properly, the resulting parent is based on read link status.

Apollo3zehn, 2017-12-14:
Sorry I think was a little bit unclear. The result I get when `ecx_config_init` finishes is always:

- Master (0)
  - context->slavelist[0].parent = 0
- AnybusXGateway (1)
  - context->slavelist[1].parent = 0
- EK1100 (2)
  - context->slavelist[2].parent = 1
- EL1008 (3)
  - context->slavelist3].parent = 2
- EL2008 (4)
  - context->slavelist[4].parent = 3

But in my opinion this is useless because the actual hierachy is:

- Master
  - AnybusXGateway
  - EK1100
    - EL1008
    - EL2008

Therefore I would expect the result to be:

- Master (0)
  - context->slavelist[0].parent = 0
- AnybusXGateway (1)
  - context->slavelist[1].parent = 0
- EK1100 (2)
  - context->slavelist[2].parent = 0
- EL1008 (3)
  - context->slavelist3].parent = 2
- EL2008 (4)
  - context->slavelist[4].parent = 2

I debugged through the code I have not the impression that this is an HW issue. To fully understand this, I hope you can give a hint how the `parent` field is actually defined in SOEM. 

I have now my own definition (and code) to work around this issue, but maybe it causes problems in other setups. To allow a slave to be a parent my conditions are:
- the parent must have at least 3 of 4 ports available
- the slave is connected to the parent´s EBUS port

nakarlsson, 2017-12-14:
I thought it was a HW problem cause I didn't expect the output you presented, now I understand. 

It is used internally for DC calculations, and should be accepted as is. Sure you can alter it if you not use function that use parent.