OpenEtherCATsociety/SOEM #144: ec_slave[slave].state operational when slave is lost

kajusK, 2018-01-09:
Hi,
I've been fighting with the simple_test example on windows. There are several issues when running setup with several slaves:

- expected WKC calculation - when using 2 slaves (with both inputs and outputs configured), expected WKC is calculated as 6, real WKC returned during communication is 5. I'm pretty new to ethercat, but shouldn't be the correct calculation of expected wkc like:
```C
 (ec_group[0].outputsWKC*2 - 1)+ec_group[0].inputsWKC;
```

Anyway, that's not a blocker, the more serious issue I hit is state when a slave is lost (disconnected). Error detection works as expected when I disconnect slave directly connected to the master, but when I disconnect a different slave, the ec_slave[slave].state remains EC_STATE_OPERATIONAL for all connected slaves, any idea why? simple_test starts printing `OK : all slaves resumed OPERATIONAL.`

Thanks,
Jakub

nakarlsson, 2018-01-09:
About WKC, check chapter 2.4 in https://download.beckhoff.com/download/document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i3.pdf. I guess you execute LRW and would get a read & write +3 giving 2 *3 = 6.

Are you running linux/simple_test/simple_test.c?

nakarlsson, 2018-01-09:
Could you provide a console log snippet and wireshark log when it happens?

kajusK, 2018-01-11:
Hi,
aha, I found the issue with wkc, the second slave was not initialized correctly, now it works as expected. Thank you.

Yes, I'm running linux/simple_test/simple_test.c. Although I running on windows due to the project requirements (I'm linux only user otherwise), the linux version works just fine. I've also tried to compile win32 version and run simple_test again, but with the same result.

The requested logs are below. First, the second slave was disconnected, reconnected and then the same procedure happened with the first slave.
 
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1621754/slaveinfo.txt)
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1621757/simple_test.txt)
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/1621758/wireshark.zip)

nakarlsson, 2018-01-11:
I'm out of office but I think I have an idea what might be the trouble, can you please try the following fix, I haven't had a chance to compile or test so bear with me. I think the issue is that we miss the "islost" due to the fact that the other slave(s) is/are still up and running as expected and trigger a bug in readstate, readstate should identify that one slave is missing and try to read the states individually.

[ethercatmain.zip](https://github.com/OpenEtherCATsociety/SOEM/files/1622830/ethercatmain.zip)

kajusK, 2018-01-15:
Thank you for your fast reply. The fix you sent works perfectly.

nakarlsson, 2018-01-15:
Thanks for testning, I’ve to make this official and use this Issue.

closed 2018-02-01 by nakarlsson