OpenEtherCATsociety/SOEM #146: SOEM overwrites outputs temporarily between ec_send_processdata() and ec_receive_processdata()

tecodrive, 2018-01-09:
In my realtime task I update my process data ( ec_send_processdata() and ec_receive_processdata() ).

If I change PDO-Outputs not in my realtime task asynchronously, some output data will send with zero. 

I have found that between sending and receiving a change in the output data are not accepted. If I set before sending the outputs, the data will remain. 

I assumed that the output data in the IOmap would not be touched or changed.

mheden, 2018-01-10:
Yes, the complete IOmap is updated on when doing a receive so you need to add a locking mechanism. There should probably be an option to only update the inputs.

tecodrive, 2018-01-11:
I've study the code for ec_send_processdata and ec_receive_processdata, It's a bit tricky for me to do the right thing. It use memcopy (byte), so the most access ist to trigger the Controlword and check the Statusword of an FU. It's a bytes access. A locking mechanism in a realtime task, i try to avoid.

nakarlsson, 2018-02-02:
You can test the latest addition for overlapping IOmap, IO will not overwrite each other

nakarlsson, 2018-02-23:
No feedback so I’ll close it

closed 2018-02-23 by nakarlsson

tecodrive, 2018-04-15:
Excuse me, I would be abroad for a few weeks and just came to test your overlap implementation.

The overlap function works great :-) 

Big thank you for your effort. This greatly simplifies synchronizing the outputs of two accessing threads.

tecodrive, 2018-04-15:
Are there differences in the ethercat frame size?

There are differences in the IOmap positions of inputs (Ioffset) & outputs (Ooffset) by using ec_config_map() or ec_config_overlap_map(). 

<pre>
ec_config_overlap_map() => IOmap =>
 
   Slave  Name         HasDC   Address   Manufactor   ID          Rev    IOmap =>  Ioffset  Istartbit  Ibytes  Ibits    Ooffset  Ostartbit  Obytes  Obits
       0                   0      0000   0x00000000   0x00000000  0x00000000             -          -       -      -          -          -       -      -
       1  EK1100           1      1001   0x00000002   0x044c2c52  0x00120000             -          -       -      -          -          -       -      -
       2  EL1002           1      1002   0x00000002   0x03ea3052  0x00120000            78          0       0      2          -          -       -      -
       3  EL5101           1      1003   0x00000002   0x13ed3052  0x04000000            79          0       6     48          1          0       4     32
       4  EL5101           1      1004   0x00000002   0x13ed3052  0x04000000            85          0       6     48          7          0       4     32
       5  EL5001           1      1005   0x00000002   0x13893052  0x03fb0000            91          0       5     40          -          -       -      -
       6  EL1008           1      1006   0x00000002   0x03f03052  0x00120000            96          0       1      8          -          -       -      -
       7  EL2008           1      1007   0x00000002   0x07d83052  0x00120000             -          -       -      -         19          0       1      8
       8  EL2008           1      1008   0x00000002   0x07d83052  0x00120000             -          -       -      -         20          0       1      8
       9  EL1002           1      1009   0x00000002   0x03ea3052  0x00120000            99          0       0      2          -          -       -      -
      10  EL3054           1      100a   0x00000002   0x0bee3052  0x00100000           100          0       8     64          -          -       -      -
      11  EL3204           1      100b   0x00000002   0x0c843052  0x00120000           108          0      16    128          -          -       -      -
      12  EL4014           1      100c   0x00000002   0x0fae3052  0x00140000             -          -       -      -         46          0       8     64
      13  ServoOne         1      100d   0x00000016   0x00020001  0x00010000           132          0      14    112         54          0      10     80
      14  ServoOne         1      100e   0x00000016   0x00020001  0x00010000           146          0      14    112         68          0      10     80

      Sum => Ibits: 564 , Ibytes: 70 , Obits: 304 , Obytes: 38

ec_config_map() => IOmap =>

   Slave  Name         HasDC   Address   Manufactor   ID          Rev    IOmap =>  Ioffset  Istartbit  Ibytes  Ibits    Ooffset  Ostartbit  Obytes  Obits
       0                   0      0000   0x00000000   0x00000000  0x00000000             -          -       -      -          -          -       -      -
       1  EK1100           1      1001   0x00000002   0x044c2c52  0x00120000             -          -       -      -          -          -       -      -
       2  EL1002           1      1002   0x00000002   0x03ea3052  0x00120000            38          0       0      2          -          -       -      -
       3  EL5101           1      1003   0x00000002   0x13ed3052  0x04000000            39          0       6     48          0          0       4     32
       4  EL5101           1      1004   0x00000002   0x13ed3052  0x04000000            45          0       6     48          4          0       4     32
       5  EL5001           1      1005   0x00000002   0x13893052  0x03fb0000            51          0       5     40          -          -       -      -
       6  EL1008           1      1006   0x00000002   0x03f03052  0x00120000            56          0       1      8          -          -       -      -
       7  EL2008           1      1007   0x00000002   0x07d83052  0x00120000             -          -       -      -          8          0       1      8
       8  EL2008           1      1008   0x00000002   0x07d83052  0x00120000             -          -       -      -          9          0       1      8
       9  EL1002           1      1009   0x00000002   0x03ea3052  0x00120000            57          0       0      2          -          -       -      -
      10  EL3054           1      100a   0x00000002   0x0bee3052  0x00100000            58          0       8     64          -          -       -      -
      11  EL3204           1      100b   0x00000002   0x0c843052  0x00120000            66          0      16    128          -          -       -      -
      12  EL4014           1      100c   0x00000002   0x0fae3052  0x00140000             -          -       -      -         10          0       8     64
      13  ServoOne         1      100d   0x00000016   0x00020001  0x00010000            82          0      14    112         18          0      10     80
      14  ServoOne         1      100e   0x00000016   0x00020001  0x00010000            96          0      14    112         28          0      10     80

      Sum => Ibits: 564 , Ibytes: 70 , Obits: 304 , Obytes: 38

</pre>

nakarlsson, 2018-04-16:
Yes thats is correct, that was the intention