OpenEtherCATsociety/SOEM #148: SDO read and write times

LeGreatHub, 2018-01-17:
Thanks for writing a great library for implementing EtherCAT in various projects.
In my use I have come across some behavior that I can't explain the cause for and I hope someone here can answer my question.

I am running SOEM on a Linux host with a single slave based on an Infineon XMC contrller and they are communicating using CoE with both PDO and SDO data.

When running in redundant mode everything is fine.
When the redundancy is lost everything slows down, especially SDO traffic is affected, calling ecx_SDOread or ecx_SDOwrite takes over 10 milliseconds on 4 byte requests.

So my question is, what times should I expect for a call to ecx_SDOread and ecx_SDOwrite, assuming a single slave that answers without any processing time?

nakarlsson, 2018-01-17:
Hard to say without having access to the system and proper diagnostics data, try to measure the penalty for sending a complete PDO frame when redundancy is lost. Since a SDO read/write consists of several read/writes the penalty is a multiple of the penalty for the PDOs.

nakarlsson, 2018-01-18:
I think you'll find the answer in ecx_waitinframe_red, it will wait for 2 frames to arrive (one sent on the primary NIC and one on the secondary NIC) since that’s how SOEM implements redundancy. When using a single slave the frame sent on the "lost" interface never arrive and you'll hang there for timeout (us). 
You could try the following to prove it, if you would have 2 slaves and "lost" the connection between the 2 slaves, you shouldn't experience the same performance loss.

So to solve it you would need to handle the timeout? Either by handling the NIC link loss? Or save the information that send failed?

LeGreatHub, 2018-01-18:
Thank you for the quick replies.

I have been timing various parts of libSOEM and you are right in ecx_waitinframe_red ends up timing out causing the delay I have experienced.

Isn't the timeouts limited to cases where it is the connection on one of the redundant interfaces on the master that is "lost"?
If multiple slaves are connected and the connection between two of them is lost, the slaves on the 2 lines should send a reply back to the master before the timeout, or have I missed something here?
I will test the above next week, just to ensure I know what happens in our system. 

My current results have taught me how difficult it can be to know how long it takes to perform a SDO read or write, the data has to pass through a lot before it has gone from a piece of software to the slave and back again.
I have tried running some small tests using the OS build in timer, yes I know it's not great but at least I'm running this in a real time OS, my test setup had had a slave and a master connected directly to each other in a redundant loop.
I found a call to ecx_SDOread would result in two read and one write operation, this could be done in less than 1 millisecond, but there was a lot of fluctuation in the time, especially when reading the slave response, so most of the time it would take more than 1 millisecond.

nakarlsson, 2018-01-18:
Yes, the timeout should be limited to those  cases, that was what I tried to explain.

If you run SSC, isn’t the mailbox communication polled there ? then you get a jitter?

LeGreatHub, 2018-01-22:
Hello

Sorry, I was in a bit of a hurry and should have spend more time reading your answer from the 18th.

I will try spending some time looking at handling the loss of an interface on a master and write to you if I find an awesome solution.

Thank you for the replies, I got the information I was seeking, if you do not have I further comments I'll close this issue.

nakarlsson, 2018-01-22:
no further comments

closed 2018-01-22 by LeGreatHub