OpenEtherCATsociety/SOEM #149: Connecting to CX8010 fails

mvukov, 2018-01-17:
Hello, I am trying to communicate with CX8010 PLC over EtherCAT using SOEM. We attached some IO to the PLC, and there should be 39B input PDO data and 32B PDO data -- this is what we see using TwinCAT. Here is the output from `slaveinfo` app:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx00051bd15b43 succeeded.
1 slaves found and configured.
Calculated workcounter 1
Not all slaves reached safe operational state.
Slave 1 State= 1 StatusCode=   0 : No error

Slave:1
 Name:CX8010 EtherCAT slave
 Output size: 0bits
 Input size: 16bits
 State: 1
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000002 ID: 1f4a6032 Rev: 00110000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1200 L: 512 F:00010022 Type:2
 SM2 A:1a00 L:   0 F:00000064 Type:3
 SM3 A:1400 L:   2 F:00000020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 07
 CoE details: 37 FoE details: 00 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```
Do I need to manually assign number of bytes for PDOs? I'd appreciate any hints :) Many thanks in advance!

nakarlsson, 2018-01-17:
SOEM rely on the PDO information being present in either the Object Dictionary or the EEPROM.

nakarlsson, 2018-02-08:
I’ll close since no feedback

closed 2018-02-08 by nakarlsson

mvukov, 2018-02-10:
Sorry, it's quite busy here. We'll investigate this further, and use before all `-map` option with slaveinfo. Perhaps we need to configure the PLC correctly to see what we want in PDOs :).

mvukov, 2018-04-06:
Hi, I dig a bit further into this. Here is what `slaveinfo <eth> -map` gives me:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp4s0 succeeded.
1 slaves found and configured.
Calculated workcounter 0
Custom hook for slaves...
Found CX8010 EtherCAT slave at position 1
CX8010: custom setup!
CX8010 slave 1 set, retval = 2
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:CX8010 EtherCAT slave
 Output size: 32bits
 Input size: 168bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000002 ID: 1f4a6032 Rev: 00110000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1200 L: 512 F:00010022 Type:2
 SM2 A:1a00 L:   4 F:00000064 Type:3
 SM3 A:1400 L:  21 F:00000020 Type:4
 FMMU0 Ls:00000000 Ll:   4 Lsb:0 Leb:7 Ps:1a00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000004 Ll:  21 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 07
 CoE details: 37 FoE details: 00 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x10
  [0x0002.0] 0x7000:0x02 0x10
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0004.0] 0x6000:0x01 0x08
  [0x0005.0] 0x6000:0x02 0x08
  [0x0006.0] 0x6000:0x03 0x10
  [0x0008.0] 0x6000:0x04 0x10
  [0x000A.0] 0x6000:0x05 0x10
  [0x000C.0] 0x6000:0x06 0x10
  [0x000E.0] 0x6000:0x07 0x08
  [0x000F.0] 0x6000:0x08 0x08
  [0x0010.0] 0x6000:0x09 0x08
  [0x0011.0] 0x6000:0x0A 0x20
  [0x0015.0] 0x6000:0x0B 0x20
End slaveinfo, close socket
End program
```
For the reference I also put slave info app output with `-sdo` option: [cx8010_sdos.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1883649/cx8010_sdos.txt).

I also checked what TwinCAT is doing and found this:

[image]

Following that, I wrote `PO2SOconfig`-compatible configuration callback:
```C++
int CX8010setup(uint16 slave) {
  int retval = 0;
  printf("CX8010: custom setup!\n");

  uint16 map_1c12[2] = {0x0001, 0x1600};
  uint16 map_1c13[2] = {0x0001, 0x1a00};

  retval += ec_SDOwrite(slave, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
  retval += ec_SDOwrite(slave, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);

  while(EcatError) printf("%s", ec_elist2string());

  printf("CX8010 slave %d set, retval = %d\n", slave, retval);
  return 1;
}
```
Now, I don't know how to handle AoE-protocol message, nor if that is needed at all.

Do you maybe have other suggestions where to look at next? Thanks in advance.

mvukov, 2018-04-14:
@nakarlsson Do you have an idea what's going on and how could I fix the situation? Thanks in advance.

nakarlsson, 2018-04-15:
From the PO2SO hook try writng the PDO in 1c12/1c13 subindex 1 instead, write the subindex count in subindex 0.

1 . Clear 1c12/13 subindex 0, =0
2. Write the PDO in subindex 1, = 0x1a00/0x1600.
3. Write the count in 1c12/13 0, =1

mvukov, 2018-04-20:
@nakarlsson Thanks for the answer. The ref manual is not quite helpful, but will continue digging.

mvukov, 2019-09-10:
Hi @Pieterjan-C Unfortunately, I haven't had more time to invest in this. On our end the idea was to implement some safety software in the PLC (CX8010) and not in the main control board. Now we do everything on the main board and don't use the PLC at all...

I was suspecting that maybe there were some setup timeouts that needed to be extended, but didn't have time to further investigate.

@ArthurKetels Do you have any suggestions?