OpenEtherCATsociety/SOEM #150: Starting a Python wrapper project

bnjmnp, 2018-01-18:
Hello, I'm about to write an open source Python wrapper for the SOEM library.

I know Python isn't a good choice for real-time applications. But I think in some situations the wrapper can be very helpful.

The plan is to add the SOEM sources as git submodul to the wrapper project (repo). To be more precise I need to create a fork with minor modifications and use this fork as git submodul. SOEM (forked) sources and wrapper sources are compiled into the wrapper binary.

When it comes to licensing the wrapper project I'm not very confident. SOEM uses a modified GPLv2 licence. Dose this modification levering the GPLv2 copyleft, and allows me to put the wrapper project under GPLv3 licence or a more typical Python licence like MIT, BSD or Apache license?

In addition to hosting the wrapper's source code on gib hub, I think about making binaries available on PyPi. I'm not sure with this. Besides realization concerns, again because all that licence stuff. Maybe you can also give me some advices on this?

nakarlsson, 2018-01-19:
We don't give advice when it comes to legal stuff.

bnjmnp, 2018-01-21:
Sorry for asking, I'm okay with this.
I appreciate your time.

closed 2018-01-21 by bnjmnp