OpenEtherCATsociety/SOEM #151: multiple slaves DC sync configuration

satishdamo, 2018-01-19:
Dear sir,
Please let me know what SOEM functions to use and in what sequence so as to synchronize multiple slaves in Ethercat through Distributed Clock functions.  For example. if connecting 3 slaves in my case and slave 2 and slave 3 both has DC options, but whether to enable DC for both these slaves and whether to set SYNC cycle time for both of these slaves or enough for just one. For note, slave 2 uses both SYNC0 and SYNC1 as its an beckhoff ovesampling encoder interface and slave 3 is an servo drive following COE ds402 profile.

Awaiting for your valuable guidance.

nakarlsson, 2018-01-19:
The answers can be found in https://github.com/OpenEtherCATsociety/SOEM/issues/142, please close issues when not active.

1. The seuquence is described several times
2. To handle multiple slaves individually, you can use the PS2SOHooks

static int slave_dc_config(uint16 slave)
 {
 if slave 1
 ec_dcsync0(slave, TRUE, period_us * 1000, calc_and_copy);
elseif slave 2
 ec_dcsync01
 ...
 return 0;
 }

ec_slave[cnt].PO2SOconfig = slave_dc_config;

satishdamo, 2018-01-24:
Dear sir,
Is it possible to configure the EcMaster to be the reference clock through SOEM.  In addition please guide me how to select a desired slave as reference clock in DC.

nakarlsson, 2018-03-29:
If you can calculate/estimate the propegation delay and other jitter in your master.
Just use the first slave if you have standard setup.