OpenEtherCATsociety/SOEM #152: Detecting when slave not initialized correctly

kajusK, 2018-01-22:
Hello,
I've been fighting with slave diagnostic for a while. When the slave is not initialized by the MCU (LAN9252 on EasyCAT module), it goes to operational mode as requested, but the WKC is not matching the expected value, it seems 2 is added instead of 3 during LRW. As a result, simple_test starts printing:

OK : all slaves resumed OPERATIONAL.

Is there a way to find out which slave is not initialized?

The slaveinfo and simple_test logs are appended below together with wireshark dump.

Thank you,
Jakub

[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1652287/simple_test.txt)
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/1652288/slaveinfo.txt)
[log.zip](https://github.com/OpenEtherCATsociety/SOEM/files/1652290/log.zip)

nakarlsson, 2018-01-22:
It is very strange, from the looks of it they get configured equally and OK, but for some reason one of the slaves don’t , from the WKC =5, handle Inputs. Have you checked error counters? Is it a random behaviour?

kajusK, 2018-01-22:
It's not a random behavior, I'm able to replicate the issue any time. I'm using EasyCAT shield for testing the ethercat technology for one project, the library provided with the shield does some basic initialization of the LAN9252. When the initialization is omitted after system power up, the uninitialized shield keeps sending incorrect WKC.

Anyway, that's not a SOEM issue and I'll find the reason why it's happening myself. What I'm after is a way to identify which slave failed to initialize properly.

kajusK, 2018-01-24:
Yes, it happens even for single slave network.

Only error counter that is nonzero is PROC_ERR_COUNT, but it contains random value every power on. Will dig deeper.

nakarlsson, 2018-02-23:
Guess it is fixed

closed 2018-02-23 by nakarlsson