OpenEtherCATsociety/SOEM #154: WinPcap alternative

Cifi79, 2018-01-29:
Have you ever consider to try an alternative to WinPcap library?
I see that there are some example here:
https://docs.microsoft.com/it-it/windows-hardware/drivers/network/ndis-core-functionality2

Do you think it could be useful for SOEM?

Thanks

nakarlsson, 2018-01-29:
It would be useful indeed, at least as an alternative,given that there are benefits from using? Interested to help out and add it to SOEM? Or anyone else?

Cifi79, 2018-02-02:
Unfortunately I am not very experienced to do this kind of modification. It could be a tip for more experienced users. I also saw that a user that fork SOEM uses NPcap instead of WinPcap, you or someone else has ever tried it?

nakarlsson, 2018-02-08:
Not that I’m aware of, the beuty with WinPCAP/Wireshark is tjatig works well on any Windows machine, if we can get that and better performance it is very interesting. Have/Can you try NPcap?

What issues do you have with winpcap?

Cifi79, 2018-02-13:
Well, I will try NPcap ( as I understood it’s compatible with WinPcap so inside SOME I don’t have to change nothing) then I’ll write to tell if is working.
With Win10 we notice two strange issue:

  1.  We have to disable “Intel Access Filter” inside network properties
  2.  If we access to the Network properties page, the SOME stop to work.

Cifi79, 2018-03-16:
Hi nakarlsson, I'm working with NPcap installed in my pc and all is working well.
Regarding the use of NDIS functionality in Micorsoft library, I think that before is better to skill myself on SOEM and after, maybe, I'll try that library.
So if you want you can close this issue.
Thanks

closed 2018-03-16 by nakarlsson