OpenEtherCATsociety/SOEM #160: EL7211 - Invalid DC sync configuration?

wlkrmrtz, 2018-02-10:
[WiresharkLogs.zip](https://github.com/OpenEtherCATsociety/SOEM/files/1713468/WiresharkLogs.zip)
Hello, I'm using the SOEM library on a raspberry pi 3 with the rt_preemt patch in order to control a servo motor, that is connected to an EL7211-0001 (Beckhoff) servo terminal. The EL7211-0001 is connected to an EK1100 coupler terminal. 
I tried to configure the distributed clock by adding "ec_dcsync0(...)" or "ec_dcsync01(...)" with the PO2SOconfig method hook to the code of red_test.c. 
Following the suggestions made in several other issues regarding the distributed clock synchronization
also didn't solve the problem. 
When I run the program, the error "Slave 2 State=0x12 StatusCode=0x0030 : Invalid DC SYNC configuration" occurs. I also noticed, that the EL7211-0001 does not reach SAFE_OP but stays in PRE_OP. Controlling other slaves like the EL1008 and the EL2008 works fine. 

I also tried to control the EL7211 using an windows pc with the EtherCat configurator of Beckhoff and it worked, regardless of whether sync1 was activated or not. 
I logged the EtherCat packets of the EtherCat startup sequence on windows and the raspberry pi (wiresharkrp, wiresharkwin) but i don't know how to discover what is wrong with the configuration of the slave in my code on the raspberry pi. The manual of EL7211 does not tell much about its configuration when TwinCat is not used. 

I'd be really thankful if someone could help me configure the slaves distributed clock correctly.

## Code:
`int slave_dc_config(uint16 slave)
{		
		//ec_dcsync01(slave, TRUE, 1000000U, 1000000U, 0);			
		ec_dcsync0(slave, TRUE, 1000000U, 0);
		return 0;
}

void redtest(char *ifname)
{
   int cnt, i, j, oloop, iloop;

   printf("Starting test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {

      printf("ec_init on %s succeeded.\n",ifname);
      
      if (ec_config_init(FALSE)>0)
      {		
		  
		  ec_configdc();
		  
		  if(ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE*4) == EC_STATE_PRE_OP)
		  {
				printf("All Slaves in PRE_OP\n");
		  }	
		  
		  for(cnt = 1; cnt <= ec_slavecount; cnt++)
		  {
				if(ec_slave[cnt].hasdc > 0)
				{
						ec_slave[cnt].PO2SOconfig = slave_dc_config;					
				}				  
		  }
		  
 
		  if ( ec_config_map(&IOmap) > 0 )
		  {
			 
			 printf("Slavezahl: %d iomap found and configured.\n",ec_slavecount);
			 
		     ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE);`

## Output of the program:
Starting Redundant test
ec_init on eth0 succeeded.
All Slaves in PRE_OP
Slavezahl: 2 iomap found and configured.
Slave:1 Name:EK1100 Output size:  0bits Input size:  0bits State: 4 delay:0.1
         Out:00000000,   0 In:00000000,   0
Slave:2 Name:EL7211-0001 Output size: 48bits Input size: 48bits State:18 delay:155.1
         Out:000420f0,   6 In:000420f6,   6
Calculated workcounter 3
Request operational state for all slaves
Not all slaves reached operational state.
Slave 2 State=0x12 StatusCode=0x0030 : Invalid DC SYNC configuration
Request safe operational state for all slaves
End redundant test, close socket
End program

nakarlsson, 2018-02-10:
Check the commsisioning parts in 6.2.3 Settings in the CoE register and specially StartUp. In the PS2SO hooks you can configure in the same way as you have working in TwinCAT.

wlkrmrtz, 2018-02-10:
So there's nothing wrong with the way the distributed clock is configured? I just have to configure the slave according to 6.2.3 in the manual?

nakarlsson, 2018-02-11:
Check the CoE StartUp in TwinCAT and check the slave 0x1C32 parameter. There you should have your answers.

wlkrmrtz, 2018-02-12:
Thanks a lot. I added the SDOwrite lines in my code to configure the EL7211 the  same way as in TwinCat using the PS2SO 
hooks, but the invalid dc sync error still occurs. I tried using SDOwrite or the `ec_dcsync0 `command. 
In the advanced settings of the DC tab in TwinCat SYNC1 is enabled, so i tried `ec_dcsync01`. That led to the fatal sync error. 
Startup:
[image]
0x1C32:
[image]

Code:
`
 int slave_dc_config(uint16 slave)
 {		
		printf("%d \n", slave);

		int wkc = 0;				
		
		//Set Output/Input PDOs
		//write16(slave, index, subindex, value)
		
		wkc += write16(2, 0x1C12, 0x01, 0x1600);
		wkc += write16(2, 0x1C12, 0x02, 0x1601);		
		
		wkc += write16(2, 0x1C13, 0x01, 0x1A00);
		wkc += write16(2, 0x1C13, 0x02, 0x1A01);
		
		//Config Slave
		//MAX CURRENT
		wkc += write32(2, 0x2003, 0x11, 0x00002710);
		
		... 
		
		//Nominal DC Link Voltage 
		wkc += write32(2, 0x2002, 0x19, 0x0000BB80);
		
		//-------- Config of DC SYNC -----------------------------------
		
		//This --> leads to StatusCode=0x0030: Invalid DC SYNC configuration
		ec_dcsync0(2, TRUE, 1000000U, 0);	
		
		//This --> leads to StatusCode=0x002c: Fatal Sync Error
		ec_dcsync01(2, TRUE, 1000000U, 1000000U, 0); 
		
		//Set DC snyc using SDO?? -->0x0030: Invalid DC SYNC configuartion						
		wkc += write16(2, 0x1C32, 1, 0x0003);
		wkc += write16(2, 0x1C33, 1, 0x0003);
		wkc += write32(2, 0x1C32, 2, 0x000F4240);
		
		printf("SDO_WCK: %d", wkc);
			
		return 0;
}

void redtest(char *ifname)
{
   int cnt, i, j, oloop, iloop;

   printf("Starting Redundant test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {

      printf("ec_init on %s succeeded.\n",ifname);
      
      if (ec_config_init(FALSE)>0)
      {		
		  
		  ec_configdc();
		  
		  if(ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE*4) == EC_STATE_PRE_OP)
		  {
				printf("All Slaves in PRE_OP\n");
		  }	
		  
		  //Slave 2: EL7211-0001
		  for(cnt = 2; cnt <= ec_slavecount; cnt++)
		  {
				if(ec_slave[cnt].hasdc > 0)
				{
						ec_slave[cnt].PO2SOconfig = slave_dc_config;					
				}				  
		  }
		  
 
		  if ( ec_config_map(&IOmap) > 0 )
		  {`

wlkrmrtz, 2018-02-13:
I got the problem fixed by `dc_sync01(slave-no of EL7211, 1000000U, 0U, 0U);`

closed 2018-02-13 by wlkrmrtz