OpenEtherCATsociety/SOEM #161: LAN9252 Doesnt reach operationl state

prabhakarlad, 2018-02-12:
Hello,

I have been trying to get SOEM working with lan9252 slave controller,  looking at the eeprom I get correct output as follows:

> ./eepromtool  eth1 1 -i                                                                                         
> SOEM (Simple Open EtherCAT Master)
> EEPROM tool
> ec_init on eth1 succeeded.
> 1 slaves found.
> Slave 1 data
>  PDI Control      : 0E89
>  PDI Config       : CC84
>  Config Alias     : 0000
>  Checksum         : 00F9
>    calculated     : 00F9
>  Vendor ID        : 000004D8
>  Product Code     : 00009252
>  Revision Number  : 00000001
>  Serial Number    : 00000000
>  Mailbox Protocol : 000E
>  Size             : 000F = 2048 bytes
>  Version          : 0001
> End, close socket
> End program

But when I try to run simple_test, the slave doesn’t reach the operational (EC_STATE_OPERATIONAL) state at all. Is there something I am  missing ?

> ./simple_test eth1
> SOEM (Simple Open EtherCAT Master)
> Simple test
> Starting simple test
> ec_init on eth1 succeeded.
> 1 slaves found and configured.
> Slaves mapped, state to SAFE_OP.
> DC capable : 1
> Slave:1 Name:PIC32 EtherCAT MSP16BIT Output size:160bits Input size: 96bits State: 4 delay:0.1
> Request operational state for all slaves
> segments : 1 : 32 0 0 0
> Request operational state for all slaves
> Calculated workcounter 3
> Not all slaves reached operational state.
> Slave 1 State=0x04 StatusCode=0x0000 : No error
> 
> Request init state for all slaves
> End simple test, close socket
> End program

nakarlsson, 2018-02-14:
Hi,

1. We recently added support for an overlapping IOmap, in https://github.com/OpenEtherCATsociety/SOEM/pull/156, could you please try an earlier version to verify that we haven't introduced a BUG.

2. Always good to run the slave CTT to verify it is implemented correct.

prabhakarlad, 2018-02-14:
Hi,

Thanks for coming back.

I did manage to get it working SOEM now, I had revert the SSC tool to older version 5.11 to get it working. Is there some mismatch with SOEM and SSC tool version 5.12 ?

I am waiting for the licence for the slave CTT, and then will I run through the SSC  tool differences between 5.11 and 5.12.

Cheers,
Prabhakar

nakarlsson, 2018-02-14:
Don't know about SSC tools, recently we had a similar issue, https://github.com/OpenEtherCATsociety/SOEM/issues/138.
I haven't looked in detail on your input since I'd say first step is to have a CTT conformant slave.

nakarlsson, 2018-02-14:
Please close the issue since it works

closed 2018-02-14 by prabhakarlad

prabhakarlad, 2018-02-14:
Just to add when I switched  the SSC tool from 5.12 to 5.11, twincat had this issue of slave not switching to operational, but only after I did "Load PDO info from device" the slave was able to switch in OP.

arkorobotics, 2020-08-17:
Although this issue was closed years ago, we stumbled across the same problem when using SSC Tool V5.12 with the LAN9252 and found another working solution:

We had an issue where the SM2 and SM3 types would appear swapped when scanning for slave info.  SM2 type enumerated as an input (4) and SM3 type enumerated as an output (3), instead of the expected types SM2 = 3 [output] and SM3 = 4 [input]. After digging in deeper, we also noticed that the SM0 and SM1 types were swapped as well.

After running across this thread and a lot of head scratching, we finally compared the SyncManager code generated by V5.11 against V5.12. The issue we found isn't entirely obvious and the solution we came up with... works... but aren't exactly sure why. Our best guess is that it has something to do with how an array of UINT16's are parsed and sent as byte to the LAN9252.

**Using SSC Tool V5.11:**
"coeappl.c" structures and packs the SyncManager type info like this:

```
typedef struct OBJ_STRUCT_PACKED_START {
UINT16 u16SubIndex0; /**< \brief SubIndex 0*/
UINT8 aEntries[4]; /**< \brief Entry buffer (one for each SyncManager)*/
} OBJ_STRUCT_PACKED_END
TOBJ1C00;

TOBJ1C00 sSyncmanagertype = {0x04, {0x01, 0x02, 0x03, 0x04}};

OBJCONST TSDOINFOENTRYDESC    OBJMEM asEntryDesc0x1C00[2] = {
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ},
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ}};
```

**Using SSC Tool V5.12** 
"coeappl.h" structures the SyncManager type info like this:
```
typedef struct OBJ_STRUCT_PACKED_START {
UINT16 u16SubIndex0; /**< \brief SubIndex 0*/
//*ECATCHANGE_START(V5.12)*/
UINT16 aEntries[2]; /**< \brief Entry buffer (one for two SyncManagers)*/
//*ECATCHANGE_END(V5.12)*/
} OBJ_STRUCT_PACKED_END
TOBJ1C00;
```

"coeappl.c" packs the SyncManager type info like this:
```
TOBJ1C00 sSyncmanagertype = {0x04, {0x0102, 0x0304}};

OBJCONST TSDOINFOENTRYDESC    OBJMEM asEntryDesc0x1C00[] = {
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ},
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ}};
```

The solution we found is to replace the V5.12 SyncManager structures and packing code with the code generated by SSC V5.11. After doing that and recompiling, the SyncManager types enumerate correctly. 

FYI: We chose not to use the V5.11 code is because by the time we found this issue, we had already invested a good deal of effort into porting the auto-generated code from V5.12 to play nice with the STM32 libraries we are using (specifically a custom HAL for the STM32F373). The level of effort required to port the code again seemed more difficult than trying to debug and replace a few lines of code.

Hope this helps someone in the future.