OpenEtherCATsociety/SOEM #162: Inputs Mapped fetches same value after a while

prabhakarlad, 2018-02-15:
Hello,

I had been trying to use the simple_test example, Initially  at the beginning  the inputs value keeps updating properly, but after a while the value is stagnant and doesn’t change (temperature sensor and potentiometer LAN9252), If I unplug the slave and plug it back again the inputs value displayed are sane, only thing I have changed in the application  is added osal_usleep(12000); instead of osal_usleep(5000); in the loop as the slave always went into ERROR : slave 1 is in SAFE_OP + ERROR, attempting ack. and later switched back to WARNING : slave 1 is in SAFE_OP, change to OPERATIONAL.
Processdata cycle 4459, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 0f 01 00 00 e8 00 00 00 00 00 00 00 00 00 00 00 01 T:3348361958617998440

OK : all slaves resumed OPERATIONAL.

To add more when the slave goes into state where the input values are same all the time if I unplug and plug back the slave, the input values come to sane state again.

Is there I am missing something to setup, can we detect such state.

Cheers,
--Prabhakar

nakarlsson, 2018-02-15:
Run the CTT and verify the slave functionallity. Adding extra time in the processdata loop to avoid the salve going into SAFEOP + ERROR don't make sense.

The inputs should update with actual values in SAFEOP as well.

You can verify with wrieshark that SOEM and the frames agree on the actual values.

prabhakarlad, 2018-02-19:
I am still awaiting for CTT, so will close for now and will re-open later.

closed 2018-02-19 by prabhakarlad