OpenEtherCATsociety/SOEM #163: Unable to detect slaves using Windows 7

ljmarks, 2018-02-22:
I'm having some trouble getting the slaveinfo program that ships with the source code to pick up slaves on an EtherCAT network. Output is:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{965BEE73-11C5-4C67-9086-6B2CB3E22D3F} succeeded.
No slaves found!
End slaveinfo, close socket
End program
```

TwinCAT 3.1 is also running on the same PC, detecting all slaves on the network. Wireshark is also picking up EtherCAT datagrams. Am I missing something here? Any help would be appreciated.

nakarlsson, 2018-02-22:
Maybe you can post the wireshark log?
Also, try to narrow it down, make sure WinPCAP send and receive the EtherCAT frames correctly? From the description that would be my first guess, that it somehow fails.

ljmarks, 2018-02-24:
Thank you for the reply. Here's an example packet picked up by Wireshark on the relevant adapter:
[https://pastebin.com/yuwGti4B](url)

WinPcap seems to pick up frames on the adapter - the output:

> 08:56:52,858304 len:60
> 08:56:52,868354 len:686
> 08:56:52,868364 len:60
> 08:56:52,868367 len:60
> 08:56:52,868375 len:686
> 08:56:52,878332 len:686
> 08:56:52,878348 len:686
> 08:56:52,878350 len:60

etc. is produced using the following packet handler callback:

```
void packet_handler(u_char *param, const struct pcap_pkthdr *header, const u_char *pkt_data)
{
	struct tm *ltime;
	char timestr[16];
	time_t local_tv_sec;

	(VOID)(param);
	(VOID)(pkt_data);

	local_tv_sec = header->ts.tv_sec;
	ltime=localtime(&local_tv_sec);
	strftime( timestr, sizeof timestr, "%H:%M:%S", ltime);
	
	printf("%s,%.6d len:%d\n", timestr, header->ts.tv_usec, header->len);
}
```

I suppose this demonstrates that WinPcap is able to receive the EtherCAT frames. Can I ask what is the best approach to investigating whether WinPcap can **send** the EtherCAT frames correctly?

Again, thank you for taking the time to respond.

nakarlsson, 2018-02-25:
Verify that SOEM send and receive the frame, put a printf in the nicdrv.c receive.

You sent a whireshark with TwinCAT communication, make sure SOEM and TwinCAT don’t run at the same time.

ljmarks, 2018-02-25:
Indeed, TwinCAT running was the issue. Thanks

closed 2018-02-25 by ljmarks