OpenEtherCATsociety/SOEM #165: Which hardware/ Embedded Device is the best choice for some tests?

mgiaco, 2018-02-23:
Hello,

I would like to test the Slave and Master. I will build a setup with 2 slaves and with 1 master. For the Slaves I will try the XMC4800 Eval Boards. For the Master I am not sure what I should use. 

Do I need any special hardware fro the master?  

1) Can i use a BeagleBone Black Industrial?
2) Or should I use a special hardware. I found this for the Raspberry : https://www.sg-electronic-systems.com/ecommerce/en/ethernet/19-etherberry-v16.html 

many thanks

nakarlsson, 2018-02-23:
Depends since you don't need a special master HW. Any Win/Linux PC can run SOEM. For embedded devices you need to provide a port for oshw/osal.

mgiaco, 2018-02-26:
Okay many thanks.

closed 2018-02-26 by mgiaco