OpenEtherCATsociety/SOEM #167: Slave version

Cifi79, 2018-03-21:
Hi everybody, I'm using a Beckhoff slave EP7041-0002 bought in 2014 and another EP7041-0002 bought in 2017. They have different fw and hw version.
I've notice that the old one don't work with SOEM, it don't reach the OPERATIONAL status.
My code initialize the PDO configuration and set the OPERATIONAL status of the slave, that's it.

It seems that twincat can manage different version of the same slave through an xml file ( i don't know if it's the ESI/ENI file).
Can anyone understand this xml file?

`<?xml version="1.0" encoding="ISO-8859-1"?>
<EtherCATConfig xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="EtherCATConfig.xsd" Version="1.3"><Config><Master><Info><Name><![CDATA[Device 2 (EtherCAT)]]></Name><Destination>010105010000</Destination><Source>901b0e146b95</Source><EtherType>a488</EtherType></Info><MailboxStates><StartAddr>150994944</StartAddr><Count>1</Count></MailboxStates><InitCmds><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[read slave count]]></Comment><Requires>cycle</Requires><Cmd>7</Cmd><Adp>0</Adp><Ado>304</Ado><Data>0000</Data><Retries>0</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[read slave count]]></Comment><Requires>cycle</Requires><Cmd>7</Cmd><Adp>0</Adp><Ado>304</Ado><Data>0000</Data><Retries>0</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[enable ECAT IRQ]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>512</Ado><Data>0400</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear configured addresses]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>16</Ado><Data>0000</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear crc register]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>768</Ado><Data>0000000000000000</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><Transition>PI</Transition><Transition>BI</Transition><Transition>SI</Transition><Transition>OI</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear fmmu]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>1536</Ado><DataLength>256</DataLength><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear sm]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>2048</Ado><DataLength>256</DataLength><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear dc system time]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>2320</Ado><DataLength>32</DataLength><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[clear dc cycle cfg]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>2433</Ado><Data>00</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[reset dc speed]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>2352</Ado><Data>0010</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[configure dc filter]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>2356</Ado><Data>000c</Data><Retries>3</Retries></InitCmd><InitCmd><Transition>IP</Transition><BeforeSlave>1</BeforeSlave><Comment><![CDATA[en/disable second physical address]]></Comment><Requires>cycle</Requires><Cmd>8</Cmd><Adp>0</Adp><Ado>259</Ado><Data>00</Data><Retries>3</Retries></InitCmd></InitCmds></Master><Slave><Info><Name><![CDATA[Box 1 (EP7041-0002)]]></Name><PhysAddr>1001</PhysAddr><AutoIncAddr>0</AutoIncAddr><Physics>YY</Physics><VendorId>2</VendorId><ProductCode>461455442</ProductCode><RevisionNo>1572866</RevisionNo><SerialNo>0</SerialNo><ProductRevision>EP7041-0002-0024</ProductRevision></Info><ProcessData><Send><BitStart>312</BitStart><BitLength>176</BitLength></Send><Recv><BitStart>312</BitStart><BitLength>192</BitLength></Recv><Sm2><Type>Outputs</Type><DefaultSize>8</DefaultSize><StartAddress>4608</StartAddress><ControlByte>36</ControlByte><Enable>1</Enable><Pdo>5633</Pdo><Pdo>5634</Pdo><Pdo>5638</Pdo></Sm2><Sm3><Type>Inputs</Type><DefaultSize>8</DefaultSize><StartAddress>6400</StartAddress><ControlByte>32</ControlByte><Enable>1</Enable><Pdo>6657</Pdo><Pdo>6659</Pdo><Pdo>6662</Pdo></Sm3><TxPdo Fixed="1"><Index>#x1a00</Index><Name>ENC Status compact</Name><Exclude>#x1a01</Exclude><Entry><Index>#x6000</Index><SubIndex>1</SubIndex><BitLen>1</BitLen><Name>Status__Latch C valid</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>2</SubIndex><BitLen>1</BitLen><Name>Status__Latch extern valid</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>3</SubIndex><BitLen>1</BitLen><Name>Status__Set counter done</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>4</SubIndex><BitLen>1</BitLen><Name>Status__Counter underflow</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>5</SubIndex><BitLen>1</BitLen><Name>Status__Counter overflow</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x0</Index><BitLen>2</BitLen></Entry><Entry><Index>#x6000</Index><SubIndex>8</SubIndex><BitLen>1</BitLen><Name>Status__Extrapolation stall</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>9</SubIndex><BitLen>1</BitLen><Name>Status__Status of input A</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>10</SubIndex><BitLen>1</BitLen><Name>Status__Status of input B</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>11</SubIndex><BitLen>1</BitLen><Name>Status__Status of input C</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x0</Index><BitLen>1</BitLen></Entry><Entry><Index>#x6000</Index><SubIndex>13</SubIndex><BitLen>1</BitLen><Name>Status__Status of extern latch</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>14</SubIndex><BitLen>1</BitLen><Name>Status__Sync error</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x0</Index><BitLen>1</BitLen></Entry><Entry><Index>#x6000</Index><SubIndex>16</SubIndex><BitLen>1</BitLen><Name>Status__TxPDO Toggle</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>17</SubIndex><BitLen>16</BitLen><Name>Counter value</Name><DataType>UINT</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>18</SubIndex><BitLen>16</BitLen><Name>Latch value</Name><DataType>UINT</DataType></Entry></TxPdo><TxPdo Fixed="1" Sm="3"><Index>#x1a01</Index><Name>ENC Status</Name><Exclude>#x1a00</Exclude><Entry><Index>#x6000</Index><SubIndex>1</SubIndex><BitLen>1</BitLen><Name>Status__Latch C valid</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>2</SubIndex><BitLen>1</BitLen><Name>Status__Latch extern valid</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>3</SubIndex><BitLen>1</BitLen><Name>Status__Set counter done</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>4</SubIndex><BitLen>1</BitLen><Name>Status__Counter underflow</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>5</SubIndex><BitLen>1</BitLen><Name>Status__Counter overflow</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x0</Index><BitLen>2</BitLen></Entry><Entry><Index>#x6000</Index><SubIndex>8</SubIndex><BitLen>1</BitLen><Name>Status__Extrapolation stall</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>9</SubIndex><BitLen>1</BitLen><Name>Status__Status of input A</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>10</SubIndex><BitLen>1</BitLen><Name>Status__Status of input B</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>11</SubIndex><BitLen>1</BitLen><Name>Status__Status of input C</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x0</Index><BitLen>1</BitLen></Entry><Entry><Index>#x6000</Index><SubIndex>13</SubIndex><BitLen>1</BitLen><Name>Status__Status of extern latch</Name><DataType>BOOL</DataType></Entry><Entry><Index>#x6000</Index><SubIndex>14</SubIndex><BitLen>1</BitLen><Name>Status__Sync error</Name><DataType>BOOL</DataType></En [... cut ...] f 1. EtherCAT command
0x0002 = copy data with wrong working counter of 2. EtherCAT command
0x0004 = copy data with wrong working counter of 3. EtherCAT command
...
0x4000 = copy data with wrong working counter of 15. EtherCAT command
]]></Comment><DataType>UINT</DataType><BitSize>16</BitSize><BitOffs>12176</BitOffs></Variable><Variable><Name>Outputs.DevCtrl</Name><DataType>UINT</DataType><BitSize>16</BitSize><BitOffs>12272</BitOffs></Variable></Outputs></ProcessImage></Config></EtherCATConfig>
`

mheden, 2018-03-22:
What state do you end up in and what is the AL status code?

Cifi79, 2018-03-22:
With the old one I end up in PreOperational with AL status code: Invalid input configuration.
With the new one I end up in Operational.

Apollo3zehn, 2018-03-25:
I have a similar issue with a very old EL6731-0010 compared to a new EL6731-0010. The new one works, the old one does not. There are at least two issues with that old slave:

- the slavelist[index].CoEdetails field of SOEM says that the slave does not support Complete CoE Access (CA) access but it rejects all download requests when trying to use legacy access (you can see it with Wireshark and ecat_mailbox filter). Therefore SOEM fails to configure it properly. When forcing to use CA instead, the second issue appears:
- before going into SafeOp mode, SOEM uploads the register contents 0x1C12  and 0x1C13 to get the number of configured PDOs. The **new** slave provides the correct number, the **old** one returns 0 which leads to incorrect slave configuration.

This does not happen in TwinCAT because TwinCAT is **not** uploading these register contents. Instead TwinCAT is somehow calculating the correct PDO count and therefore can configure the slave correctly.

I have seen similar behavior also with an old EL6601 and currently I am investigating a new EL6751. For more details see [this issue](https://github.com/OpenEtherCATsociety/SOEM/issues/120). For me it seems that some old slaves are only providing proper data via CoE after they reached SafeOp instead of PreOp, which could be a bug. When I have more time I will continue investigating this. Right now I am not sure if it is really faulty behavior or if I do a mistake but maybe this gives you a hint. 

Of course it could simply be that the old version of your slave just needs to be configured differently.

nakarlsson, 2018-05-11:
Can this issue be closed?