OpenEtherCATsociety/SOEM #168: Cannot configure the syncmode into FreeRun

ErwinLu, 2018-04-12:
Dear developers，

Thanks a lot for your contribution of writing such a great library.

I am a newbie to EtherCAT trying to run a SANYO servo motor(a simple task) by controlling its servo driver(win10; SANYO RS2 servo driver; the servo uses CoE). 
What I am doing is modifying simple_test.c so that I can write control words(0x6040)/read status words(0x6041) and some other OD entries through ec_SDOWrite()/ec_SDORead() funcs after `inOP = TRUE;.`
When runs at default SM2 mode, the program is able to change the OD entries above occasionally, which means it fails most times. An SANYO engineer suggest that I should run the program in FreeRun mode, so I add
`unit16 syncmode;
int length;
syncmode = 0x00;  //means freerun mode according to the servodriver manual
length = sizeof(syncmode);
ec_SDOwrite(1, 0x1c32, 0x01, FALSE, length, &syncmode, EC_TIMEOUTRXM);`
after `ec_init(ifname)` and before `ec_config_init(FALSE)`.
It turned out that the mode could be changed into freerun sometimes but most times it just failed and remained SM2 mode.

I am confused, is simple_test.c able to be configured into FreeRun mode at all? If can, then what steps did I miss?

I have looked through all the issues yet I did not find a similar one(maybe I missed something).
Sorry if my question is too basic but I have been stuck for days and I really need your help.
Thank you for your time.

ErwinLu, 2018-04-16:
Thanks for your reply.
By the way, how often do ec_send_processdata() and ec_receive_processdata() work? Is it up to the running time of per RTthread()? And what is the relationship between RTthread()'s interval and the cycle time of the slava(e.g., 8ms)?

nakarlsson, 2018-04-16:
The send/recv cycle is the cycletime. In free-running it is up to the application how fast you want to run. How often you can call send/recv depends on the overhead in the system and frame size.

ErwinLu, 2018-04-25:
Thx for your reply. 
Though was not able to change into freerun, I changed the program into SYNC0 and my problem is partly solved. So I will close the issue.
Thank you again for your contribution.

closed 2018-04-25 by ErwinLu