OpenEtherCATsociety/SOEM #172: cannot detect slave

joeization, 2018-05-02:
I'm using tsino-dynatron's CoolDrive R series, but i can not detect it using soem(slaveinfo).
Twincat can detect and run correctly.

Here is the wireshark log for soem
2	1.879224	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x103, Wc 0 
3	1.879261	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x4, Ado 0x103, Wc 4 
4	1.879578	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x120, Wc 0 
5	1.880121	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x4, Ado 0x120, Wc 4 
6	1.880403	Private_01:01:01	Broadcast	ECAT	29	'BWR': Len: 1, Adp 0x0, Ado 0x120, Wc 0 
7	1.880921	03:01:01:01:01:01	Broadcast	ECAT	60	'BWR': Len: 1, Adp 0x4, Ado 0x120, Wc 4 
8	1.881024	Private_01:01:01	Broadcast	ECAT	30	'BRD': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
9	1.881750	03:01:01:01:01:01	Broadcast	ECAT	60	'BRD': Len: 2, Adp 0x4, Ado 0x0, Wc 4

Thanks

mheden, 2018-05-02:
4 slaves responds according to your log excerpt. What indicates that your slave doesn't respond?

joeization, 2018-05-02:
slaveinfo output that 0 slave found

nakarlsson, 2018-05-02:
Make sure TwinCAT is stopped, otherwise it will consume the frames before they SOEM?

joeization, 2018-05-02:
twincat was deactivated, i also tried soem on another laptop(no twincat installed) but still not working

nakarlsson, 2018-05-02:
Ok, can you verify that SOEM receive the frame response? Add a counter , print or anything.

joeization, 2018-05-03:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{} succeeded.
ecx_BWR wkc = 0
ecx_BWR wkc = 0
ecx_BWR wkc = 0
ecx_BRD wkc = 0
ecx_detect_slaves wkc = 0
ec_config_init wkc = 0
No slaves found!
End slaveinfo, close socket
End program

also test on more laptops
2 of 5 can find cool drive and configure, maybe the problem is on adapter?

nakarlsson, 2018-05-03:
We use WinPCAP and I’ve run on several built in PCIE NICs and USB NICs without problem, so you need to investigate where the frames end up.

Apollo3zehn, 2018-05-10:
Could it be a firewall issue? Wireshark receives all frames before they reach the firewall. SOEM is also based in WinPCAP, so it should also not be affected. But it could be worth a try to disable it temporarily.

joeization, 2018-05-10:
disable firewall cannot find slave, and Wireshark log still same

nakarlsson, 2018-05-10:
Creat a simple EtherCAT frame, send it with pcap_send. Use pcap_recv to verify you get it back. If no frame is returned its difficult for us to say what it can be. 

Again, make sure no TwinCAT service is running.

accrea, 2018-05-17:
I've encountered similar problems with WinPcap and Win10Pcap when running SOEM on Windows10. Everything started to work properly after installation of Npcap https://nmap.org/npcap/ Sorry if it's not relevant.

nakarlsson, 2018-05-17:
Ncap sound interesting regardless

joeization, 2018-05-18:
Npcap works but i am not sure what's going on.
and my master program has some errors, needs to do several tests

closed 2018-07-06 by joeization