OpenEtherCATsociety/SOEM #173: Can anyone explain response generated from SOEM Master(simple_test)?

swapnil-mali, 2018-05-26:
Here is response after running simple_test:
_sudo ./simple_test eth1
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth1 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 4 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle    2, WKC 3 , O: 31 cc bb aa I: 00 00 00 00 T:3350946613939947480
OK : all slaves resumed OPERATIONAL.
MESSAGE : slave 1 reconfigured, O: 00 00 00 00 I: 00 00 00 00 T:3350946613998343940
Processdata cycle   15, WKC 3 , O: 31 cc bb aa I: 00 00 00 00 T:3350946614009767605
WARNING : slave 1 is in SAFE_OP, change to OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
Processdata cycle   21, WKC 3 , O: 31 cc bb aa I: 00 00 00 00 T:3350946614040739945
Processdata cycle   28, WKC 3 , O: 31 cc bb aa I: 00 00 00 00 T:3350946614076503150_

Curious to understand below responses:
1. MESSAGE : slave 1 reconfigured, O: 00 00 00 00 I: 00 00 00 00 T:3350946613998343940
2. Processdata cycle   15, WKC 3 , O: 31 cc bb aa I: 00 00 00 00 T:3350946614009767605

nakarlsson, 2018-05-26:
1. The example show how you can detect and recover slaves from errors, basically when the wkc differ from the expected or ALStatus differ from OP. In a normal system you shouldn’t se this print-out.

2. Just a status print of processdata, cycle, wkc and so on.

You can study the code to better understand.

nakarlsson, 2018-06-17:
I will close this, its pretty straight forward

closed 2018-06-17 by nakarlsson