OpenEtherCATsociety/SOEM #174: Endianness problem on PPC604 target

willytax, 2018-06-01:
I'm trying to use SOEM on a PPC604 target (MPC5200) running busybox (linux kernel 2.6.28), therefore in ethercattype.h I changed #define EC_LITTLE_ENDIAN to #define EC_BIG_ENDIAN.
I compiled the slaveinfo example code on Eclipse using a cross-compiler. 
Comparing the output that I get on the ppc machine with the one I obtain running the same code on a linux intel PC I can see that tsome endianness problems persist. For example all the addresses displayed are byte swapped and I can't get the device map correctly.
I'm missing something?
Has anyone encountered this problem jet?

Thankyou

nakarlsson, 2018-06-03:
I’m sure we have more run-time on Little endian and I’m not sure how many installations running on Big Endian at all. So what you see is what you get, feel free to lift the Big endian version of SOEM or if we’re lucky someone have the missing pieces in a patch ready.

We at rt-labs have never run SOEM on Big Endian.

nakarlsson, 2018-07-02:
Can you give some more details round this issue?

nakarlsson, 2018-11-14:
Fixes have been done in Endian fixes #222 , write a new issue if more fixes is needed

closed 2018-11-14 by nakarlsson