OpenEtherCATsociety/SOEM #175: cannot find salve

JianchengYang, 2018-06-04:
run the soem cannot find slave and add information to detect whether the program is running correctly and the information cannot be printed out，but the original information of the program can be printed out normally

nakarlsson, 2018-06-04:
Please search the Issues for similar entries, for example already open https://github.com/OpenEtherCATsociety/SOEM/issues/172.

JianchengYang, 2018-06-05:
Hello, I have checked #172， then I tried to use another computer to do the same operation, and I found the slave station. Is it related to the network card of the computer?

nakarlsson, 2018-06-06:
Seems so, you need to provide SOEM with the abillity to send layer 2 frames , the reference use WinPCAP. If that don’t work you can use another layer 2 solution.

closed 2018-06-06 by nakarlsson