OpenEtherCATsociety/SOEM #176: Windows DC Synchronisation

yoyixms, 2018-06-06:
Hi,
I am testing a Beckhoff ELM3004 with oversampling (10 samples per cycle), but it seems that due to the timing of the thread, it does not always stay synchronised, giving wrong data. well that's at least what I think, because i see that the variable No. of Samples of the ELM3004 changes from 10 to 9  for a while and and then goes back to 10. 

What would be the best way to syncronize the thread that sends and receives the EtherCAT Frame with the EtherCAT cycle time? I am currently using 10 milliseconds. is there also a minimum cycle time that would work properly on windows?

Maybe the problem is caused because of something else, but i am not really sure how to diagnose it. 

Thanks in advance for your help

Apollo3zehn, 2018-06-07:
What do you mean by changing number of samples? The IO map has always the same size. Do you mean that sometimes you have 9 real values plus a zero-value?

yoyixms, 2018-06-07:
I plotted some signals to make it a bit more clear. I also set the oversampling to 50.
[image]
as you can see, as soon as the number of samples for some unknown reason decreases, the sample [40] of the array stops changing. that does not happen by sample [0].

Apollo3zehn, 2018-06-08:
If my understanding is correct, it is not possible to access half recorded data in Ethercat. The Ethercat slave controller provides sync managers for data exchange. If buffered mode is enabled, you will always get a consistent chunk of data. If you are reading the data slower than they are written, you will loose data of the last cycle (whole dataset). I am not sure what happens if you are requesting data more offen than they are written, but I guess you just get a copy of the last written data. In none of these cases you can get a partial dataset. 

I tried to get the manual to understand the actual meaning of the variable "Status No. of samples". But it seems that Beckhoff is not yet providing it. 

Regarding synchronisation: Currently I am trying to keep in sync with Ethercat by modifying the ECT_REG_DCSYSOFFSET register regularily. I know this is not the best approach, but in older Windows versions it is not possible to get a high resolution time stamp of the absolute time. Instead the time stamp is updated with the system wake-up cycle (e.g every 1 ms or 15.6 ms, depending on your settings). So this time cannot be used reliably for the ESC time control loop. 

Windows 10 (and 8?) offer a new API to get the actual time in high resolution. I did not yet test if you can exactly synchronize Ethercat and Windows clock with this new API.

yoyixms, 2018-06-08:
Thanks for the link!
Would you give me an example showing how you regularily modify that register?

Apollo3zehn, 2018-06-08:
Here is the actual function:

https://github.com/OneDAS-Group/OneDAS-Ethercat/blob/master/native/OneDas.Ethercat.Native/OneDas.Ethercat.Native.c#L271

which is called here:

https://github.com/OneDAS-Group/OneDAS-Ethercat/blob/a544ffdad334f46426e1d33ba6a35b87c8aa8b12/src/OneDas.Extension.Ethercat/EthercatGateway.cs#L385

I will try to give a more elaborated answer in Monday.

Apollo3zehn, 2018-06-11:
The basic idea is that each IO update cycle, the programs reads out the [EtherCAT time](https://github.com/OneDAS-Group/OneDAS-Ethercat/blob/a544ffdad334f46426e1d33ba6a35b87c8aa8b12/src/OneDas.Extension.Ethercat/EthercatGateway.cs#L315) and compares it to current [Windows time](https://github.com/OneDAS-Group/OneDAS-Ethercat/blob/a544ffdad334f46426e1d33ba6a35b87c8aa8b12/src/OneDas.Extension.Ethercat/EthercatGateway.cs#L369). When the average difference is bigger than 1.5 ms, it is [compensated](https://github.com/OneDAS-Group/OneDAS-Ethercat/blob/a544ffdad334f46426e1d33ba6a35b87c8aa8b12/src/OneDas.Extension.Ethercat/EthercatGateway.cs#L385) until the delta is below 1 ms.

Because I do not change the Windows clock, the update IO method calls (`ecx_send_processdata`, `ecx_receive_processdata`) will still be in sync with the Windows time. But I think is not true anymore as soon as the time is changed by e.g NTP time synchronization or manual time modification because the OS timers just sleep for relative amount of time, ignoring the absolute time or its changes.

Therefore I do not use timers and instead let the IO thread [sleep several times](https://github.com/OneDAS-Group/OneDAS-Core/blob/c353474b803e633be341b127c12597dcfac5e327/src/OneDas.Core/Engine/RtTimer.cs#173) until the next point in time is reached. For example, with a base frequency of 10 ms, and current time of 08:45:19.022, the thread sleeps until the time is 08:45:19.030 (8x). Then again, if the IO update took 1 ms, the time is 08:45:19.031, so the thread sleeps 9x until the time is 08:45:19.040. I need to use this approach (absolute time) because I am working on a measurement data acqusition software and whichs needs to ensure that there is always the same number of samples per time period. The working principle is also described [here](http://onedas.readthedocs.io/en/latest/2_data_aquisition/3_server/6_timing.html#reference-clock) (the linked page is not up to date and also not finished, but maybe helpful).

When you just need PLC behavior, maybe it is sufficient to use relative time (e.g the number of CPU core ticks since boot-up, see [Time Stamp Counter (TSC)](https://en.wikipedia.org/wiki/Time_Stamp_Counter)). You would then take this time value, scale it properly to make it comparable to the EtherCAT time value and then compensate the EtherCAT time if necessary as shown above.

A disadvantage using the Time Stamp Counter is that only newer hardware versions support a synchronized register value between multiply CPU cores. Older version would give you different results on different cores because a core can sleep and will then count slower than normal. It is also important to mention that different versions of Windows use different hardware timers: http://www.windowstimestamp.com/description (section 2.1.4) / [Acquiring high-resolution time stamps](https://msdn.microsoft.com/en-us/library/windows/desktop/dn553408%28v=vs.85%29.aspx).

yoyixms, 2018-06-12:
Thanks for your help! I will try to do what you did ;)

nakarlsson, 2018-07-02:
Can this issue be closed?

closed 2018-07-02 by yoyixms