OpenEtherCATsociety/SOEM #178: simple_test failure

greyltc, 2018-06-08:
I've got a Beckoff EK1100 with one EL3318 thermocouple input module which I can't seem to get working and I wonder if anyone here can offer me any insights as to what's wrong and how to fix it.

`slaveinfo -map` gives:
```
$ sudo ./test/linux/slaveinfo/slaveinfo enp0s20f0u1u1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s20f0u1u1 succeeded.
2 slaves found and configured.
Calculated workcounter 1

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0
PDO mapping according to SII :

Slave:2
 Name:EL3318
 Output size: 0bits
 Input size: 256bits
 State: 4
 Delay: 150[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 00000002 ID: 0cf63052 Rev: 00150000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000024 Type:3
 SM3 A:1180 L:  32 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  32 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 07 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 210[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
End slaveinfo, close socket
End program
```
Already something seems to be wrong here because `PDO mapping according to CoE :` is blank there. Things are not much better with `simple_test`. It spams my console with ERROR and OK messages without ever printing any process data:
```
$ sudo ./test/linux/simple_test/simple_test enp0s20f0u1u1 
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp0s20f0u1u1 succeeded.
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 32 0 0 0
Request operational state for all slaves
Calculated workcounter 1
Operational state reached for all slaves.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
ERROR : slave 1 lost
MESSAGE : slave 1 found
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
ERROR : slave 1 lost
ERROR : slave 2 lost
MESSAGE : slave 1 found
MESSAGE : slave 2 found
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
ERROR : slave 1 lost
ERROR : slave 2 lost
MESSAGE : slave 1 found
MESSAGE : slave 2 found
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
...
```
Is this indicative of some problem with my network interface or in my network stack? I've tried two different USB ethernet dongles and the PCI connected ethernet port on this laptop with similar results.

The Beckhoff modules themselves are fine because I can switch them to another machine running an etherlabs setup and the etherlabs cli tools operate as expected and I can read the thermocouple temperatures fine there.

Is this a bug in SOEM?

nakarlsson, 2018-06-08:
I don’t think it is a SOEM bug, nothing points in that direction, rather instabile send/recv abillity of your laptop. A wireshark log might give you some guidance, or tracing the working counter result.

greyltc, 2018-06-08:
https://github.com/OpenEtherCATsociety/SOEM/blob/0398b6ba72f909f300ada01e92bc490b64f427cd/test/linux/slaveinfo/slaveinfo.c#L308

`ec_SDOread` gives -1 for `wkc` there

"instabile send/recv ability" of my laptop? All the other traffic through my network interfaces seems to work as expected. Is it known that SOEM doesn't work on some setups?

nakarlsson, 2018-06-09:
All your findings point to that you lose or timeout on frame receive, a wireshark would help to rule out issues sending.

So you should investigate if

1. Some frames get lost
2. You receive a frame but not within expected time limit
3. The timer used for timeout measurments are working correctly

Since this is SOEM standard code one might suspect it is some issue on your system?

The logs above show that you atleast can reach OP in some cases, that alone pointing towards a stabillity issue rather than a bug in SOEM.

greyltc, 2018-06-10:
My network interface uses the e1000e kernel module. When I `modprobe e1000e` I get the following in my journal:
```
Jun 10 14:47:46 radish kernel: e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
Jun 10 14:47:46 radish kernel: e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
Jun 10 14:47:46 radish kernel: e1000e 0000:00:1f.6: Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode
Jun 10 14:47:47 radish kernel: e1000e 0000:00:1f.6 0000:00:1f.6 (uninitialized): registered PHC clock
Jun 10 14:47:47 radish sudo[2920]: pam_unix(sudo:session): session closed for user root
Jun 10 14:47:47 radish systemd-udevd[2928]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Jun 10 14:47:47 radish kernel: e1000e 0000:00:1f.6 eth0: (PCI Express:2.5GT/s:Width x1) a4:4c:c8:28:55:c4
Jun 10 14:47:47 radish kernel: e1000e 0000:00:1f.6 eth0: Intel(R) PRO/1000 Network Connection
Jun 10 14:47:47 radish kernel: e1000e 0000:00:1f.6 eth0: MAC: 12, PHY: 12, PBA No: FFFFFF-0FF
```

Are there any known issues/interactions with SOEM and this module or this interface?

I wonder if `Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode` is my problem here. Any idea how to set that to some higher performance mode? Maybe then my timing issues here would go away.

greyltc, 2018-06-10:
Attached is a wireshark capture of `slaveinfo enp0s31f6 -map`. Unfortunately I don't have a known good capture to compare it to.
[slaveinfo_map.pcapng.gz](https://github.com/OpenEtherCATsociety/SOEM/files/2088101/slaveinfo_map.pcapng.gz)

greyltc, 2018-06-10:
I just found this note which seems extremely relevant:
https://github.com/OpenEtherCATsociety/SOEM/blob/0398b6ba72f909f300ada01e92bc490b64f427cd/drvcomment.txt#L2
But it changes nothing!

greyltc, 2018-06-10:
Multiplying all the timeout values in soem/ethercattype.h by 10 doesn't fix the issue.

ArthurKetels, 2018-06-11:
Thanks for the wireshark capture, that helps to get to the source of the problems.

First of all I see in the capture you have changed the slave stack.
EK1100 - EL3318 - EL3204 - EL3202
Try to keep the test setup equal during you questions. Otherwise other issues might sneak in undetected.

The slave discovery goes all fine. Up to the point a mailbox transfer is send (SDO request). There is no response. Not from slave 2 (el3318), nor from slave 3 and 4. All other packets are received shortly after they have been send. So you nic driver settings seem to be ok.

SOEM continues to interrogate the alternative data source, the slave's eeprom. Although it gets far and actually enough data is gathered to configure the slaves to safe-op, a lot is still missing. That is why you see no data after the "PDO mapping according to CoE".

 I have to elaborate on my "no response" remark. It is fine a slave does not react on a packet. The packet is still returned to the master but with workcounter still zero. In this case the packet is not returned at all. This could mean any of the below things:
1. Packet is corrupted during transmission from the master to the slaves
2. Packet is corrupted during return from the slaves to the master
3. Slave is marking the packet as "bad"
4. Packet is filtered by the master network stack /driver before being transmitted
5. Packet is filtered by the master network stack /driver after being received but before being put in the pcap buffer.

1 to 3 are highly unlikely because it only happens with the mailbox requests and not with any other packet. So I would rule out any hardware issue.

There is no way to tell if it is 4 or 5 from the wireshark capture. But I think there is some packet filtering going on for sure. Check by changing the master (other PC / laptop), and checking any firewall settings.

An other thing I noticed is that all packets that get through are smaller than 60 bytes (minimum ethernet packet length). Those packets are actually padded by the NIC to reach 60 bytes.

greyltc, 2018-06-12:
Hi @ArthurKetels, thanks very much for your detailed analysis. Apologies for changing the hardware on you like that. I hope it's alright that I change it one more time. I put the EL3318 in another setup.

I'll give some more details about my setup and some other things I've tried to try to address some of your potential root causes. I'm running a full gnome gui desktop on my laptop -- Arch Linux, with kernel 4.16.13. I've not purposefully installed any firewall or packet filtering software on here, and I don't believe Arch has any on by default, but I'll have a closer look at ebtables/iptables for sure.

I've tried building and running `slaveinfo` in a CentOS 7 virtual machine (again full gnome gui desktop, kernel 3.10 ) on my laptop with a usb ethernet adapter passed through; that works fine.

I've tried running `slaveinfo` in an Arch Linux virtual machine (bare bones, baseline command line only install with just enough to compile SOEM) and that does not work. So I'm suspecting it's either some bad configuration that Arch comes with out of the box or some issue caused by the newer kernel that Arch uses. So lately I've been furiously recompiling different kernels and trying those with no luck so far.

greyltc, 2018-06-12:
I've definitely got some iptables rules set up on my machine. Looks like most of them have been put there by my docker installation:
```
$ sudo iptables -S
-P INPUT ACCEPT
-P FORWARD DROP
-P OUTPUT ACCEPT
-N DOCKER
-N DOCKER-ISOLATION-STAGE-1
-N DOCKER-ISOLATION-STAGE-2
-N DOCKER-USER
-A INPUT -p udp -m multiport --dports 5353 -j ACCEPT
-A FORWARD -j DOCKER-USER
-A FORWARD -j DOCKER-ISOLATION-STAGE-1
-A FORWARD -o docker0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o docker0 -j DOCKER
-A FORWARD -i docker0 ! -o docker0 -j ACCEPT
-A FORWARD -i docker0 -o docker0 -j ACCEPT
-A DOCKER -d 172.17.0.2/32 ! -i docker0 -o docker0 -p tcp -m tcp --dport 443 -j ACCEPT
-A DOCKER -d 172.17.0.3/32 ! -i docker0 -o docker0 -p tcp -m tcp --dport 4567 -j ACCEPT
-A DOCKER-ISOLATION-STAGE-1 -i docker0 ! -o docker0 -j DOCKER-ISOLATION-STAGE-2
-A DOCKER-ISOLATION-STAGE-1 -j RETURN
-A DOCKER-ISOLATION-STAGE-2 -o docker0 -j DROP
-A DOCKER-ISOLATION-STAGE-2 -j RETURN
-A DOCKER-USER -j RETURN
```

greyltc, 2018-06-12:
`ebtables` is not on my system and some clear thinking tells me that probably `iptables` is irrelevant here.

ArthurKetels, 2018-06-12:
iptables is only related to IP packets. EtherCAT has a different protocol header 0x88a4. So only ebtables could influence non IP packets on your system. But with newer kernels and eBPF you can do almost anything with packets. So I do not rule out some special configuration at boot. With a few lines of C code I can inject packet filtering without touching any of the standard tools.

I am not a specialist in these things. But I have read about selinux doing really low level stuff here.

A way of cheking if the problem is in the incoming or outgoing direction (or both) is to connect another pc instead of the EtherCAT hardware. And then replay the wireshark capture. You can do that on both ends and capture with wireshark on the other end. You can see what comes through in both directions.

ArthurKetels, 2018-06-12:
Just found out that Arch linux supports advanced traffic control with tc
Perhaps it has some rules set to drop non IP packets?

greyltc, 2018-06-13:
Perhaps, I'm just trying to find a way to completely disable the darn thing now.

meiguwiki, 2018-06-14:
I met some similar error before.
try isolate the problem:
comment out code as following in simple_test, and if you still see the error, then should be your cable/port issue or your OS issue. otherwise, it's something else

            for(i = 1; i <= 10000; i++)
            {
              /* ec_send_processdata();
               wkc = ec_receive_processdata(EC_TIMEOUTRET);

                    if(wkc >= expectedWKC)
                    {
                        printf("Processdata cycle %4d, WKC %d , O:", i, wkc);

                        for(j = 0 ; j < oloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].outputs + j));
                        }

                        printf(" I:");
                        for(j = 0 ; j < iloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].inputs + j));
                        }
                        printf(" T:%"PRId64"\r",ec_DCtime);
                        needlf = TRUE;
                    }
              */
                  // just leave the delay in the loop
                    osal_usleep(5000);

                }

meiguwiki, 2018-06-14:
can you attach the simple_test log of wireshark?

aydal, 2018-07-06:
I have been working with SOEM and started facing this same problem just a few days ago. Every sample test was running fine before that. I am using ubuntu 16.04 and apparently, the latest kernel was giving me this problem. 
After spending hours on this I simply switched to older kernels and now it's all working fine again. I do not know if the problem is same but its worth a shot.

Stephenyu113, 2018-07-06:
Yeah, I am same with @aydal . 
When I use SOEM-1.3.1 on ubuntu 14.04 with preempt kernel 3.14, it runs well.

meiguwiki, 2018-07-07:
you are right, 
looks like some kernal has longer delay, and will cause the frame receive timeout of SOEM

aydal, 2018-07-08:
SOEM works better in RTOS systems. Xenomai gives me better results than normal Linux kernels (checked using red_test)

greyltc, 2018-07-10:
does anyone know what the most recent vanilla kernel version is that works with SOEM?

TroyChu, 2018-08-06:
Hi dude！Have you solved your problem? I'm now going through the same problem.

greyltc, 2018-08-07:
Nope, haven't solved it. I'm surprised this isn't getting more attention. Maybe they're dropping support for vanilla Linux?

TroyChu, 2018-08-07:
I'm driven crazy by SOEM...It at first worked pretty well, but some how it suddenly can't work.

meiguwiki, 2018-08-07:
Hi, guys
take a look of this one the ethtool config
https://github.com/OpenEtherCATsociety/SOEM/issues/184

nakarlsson, 2018-08-07:
We’re not planing to drop support for vanilla Linux, however, SOEM require support by the OS or bare metal application to transmitt L2 frames. Currently it seems you’re having trouble to get that support from your OS. Since you can reproduce the issues on your systems the ideal is if you could help us find a solution or atleast formulate the issue(s) needing a solution. Probably the core issue is that frames are getting dropped, but how, were and why on what systems are the keys to fix it. 
I my self rarely run SOEM on vanilla Linux.

meiguwiki, 2018-08-07:
Guys please capture the packets by wireshark and upload it.
I can help take a look.

greyltc, 2018-08-07:
@meiguwiki, do you need something  more than the wireshark capture in this comment above --> https://github.com/OpenEtherCATsociety/SOEM/issues/178#issuecomment-396068458 ?

TroyChu, 2018-08-07:
It's so wierd I can't install Wireshark on my Ubuntu ...

meiguwiki, 2018-08-07:
@greyltc wireshark capture  is enough, 
please do capture from the beginning

meiguwiki, 2018-08-07:
@TroyChu I'm ok installing wireshark on ubuntun

TroyChu, 2018-08-07:
Hi @meiguwiki Do you know why I can get data from the ethercat port, but there's no ecat data coming through that?
[image]

meiguwiki, 2018-08-07:
did you select the right port?
also did you start the simple test?

TroyChu, 2018-08-07:
Do I need to start the simple_test? The simple_test will end after a few seconds it starts.

meiguwiki, 2018-08-07:
yes, you have to start simple_test, otherwise you'll see no ecat frames

greyltc, 2018-08-09:
@meiguwiki I've already attached a wireshark capture to this thread (see my second comment from June 10th above) this capture is indeed from the beginning. Do you need something more than this?

I'm actually struggling to understand why this is a hard problem to replicate. Boot from the latest Ubuntu ISO which should have a new enough kernel and you should see it (I think). I will test this and make a complete write up here of exactly how you (or anyone else) can reproduce this issue on your own system.

meiguwiki, 2018-08-10:
@greyltc 

take a look at the highlighted frames, 
two odd things:

1. it looks to me the slave doesnt accept the coe sdo frame. 
all the coe sdo frames didnt send successfully (either no frame went back or working counter didnt count)

2. the time gap is 0.083144150 - 0.077360395 (much longer compared to others, other frame round time is around 0.0001xxx ), I suspect this should be the kernel issue, and didnt response the interrupt fast enough.

[image]

meiguwiki, 2018-08-10:
see the highlighted, 

didnt 
[image]

meiguwiki, 2018-08-10:
hey man,  found an interesting stuff. 

All  ecat ethernet frames with length longer > 60  was not send successfully, and didnt come back to the master.

ArthurKetels, 2018-08-11:
Please guys, do not hijack this thread for other issues. The issue @greyltc has is different from @TroyChu. Make your  own thread for your own question. There can be many reasons why your particular set-up does not function as expected.

ArthurKetels, 2018-08-11:
I aslo do not use vanilla linux for SOEM. It's real time behaviour is just not good enough. That aside it still should just work. 

That is why I have installed a fresh Ubuntu 18.04 on a laptop (i5 realtech NIC). Downloaded latest SOEM and build it with default settings.  This works for me with multiple different EtherCAT slaves. Both simple_test and slaveinfo have no issues. This is with stock Ubuntu kernel 4.15. 

I have also compiled a new kernel with preempt-rt 4.16. Copied the standard Ubuntu .config and only changed to preempt-rt. This also works with no problems.

I did notice a new behaviour with the ethtool. rx-usecs and tx-usecs are no longer to be set at zero. When you do that there is no or broken packet tx and rx behaviour. On my box I had to set a minimum of 3. For safety I would recommend 15 or 20 as values. Perhaps this also solves your problems.

Otherwise I would ask you specify your kernel version and nic module in use on your system. It could be that there is a issue with a specific nic driver.

MCFurry, 2018-08-27:
Hi @greyltc @ArthurKetels,

Last week I started having SOEM problems on my 16.04 system when I found this thread.
It seems that the latest kernel update to 4.15.0-33 broke SOEM functionality. 
Interestingly slaveinfo is still able to identify an EtherCAT stack correctly, but every call I do to an ec_SDOread() in my own code returns a -1 workcounter. 
simple_test also throws a lot of slave lost errors on a simple Beckhoff stack.

I briefly tried enlarging the timeout value or setting coalesce parameters for the interface which didn't help. Rebooting to an older kernel (4.15.0-32 for example) solves the problem immediately, so I'll stick to that for now.

ArthurKetels, 2018-08-27:
Now, this is very interesting, as it would make it much easier to find the change that caused this issue.
As I wrote above, on my test system with 4.16-rt there are no real problems. It still could have to do with a specific hardware configuration.

MCFurry, 2018-08-28:
I'm not sure if this kernel issue exactly relates to the original poster's question, but it might be relevant to anyone experiencing such problems anyways.

My hardware configuration is not that special, I'm using a standard Dell latitude notebook with I219-LM network card directly plugged to a standard Beckhoff stack (EK1100 plus some I/O (EL2502, EL4002, EL3102)). 

I'm sticking with the 4.15.0-30 kernel for now since it works most stable in my opinion. but if there are any relevant tests I could do on the 4.15.0-33 one, please let me know!

ArthurKetels, 2018-08-30:
I think the solution is in : https://github.com/OpenEtherCATsociety/SOEM/issues/203#issuecomment-417306167

It explains the behaviour by @greyltc  and @MCFurry. So it not so much that SOEM is broken in Linux but simply Linux is broken in certain distributions.

greyltc, 2018-08-30:
I think https://github.com/OpenEtherCATsociety/SOEM/issues/203#issuecomment-417306167 nails it. I've been on Arch Linux which tends to track the latest upstream kernel pretty closely and recently I've seen this problem go away (as if by magic), but a kernel fix would explain it.

nakarlsson, 2018-09-10:
@greyltc , can we close the issue?

greyltc, 2018-09-10:
Yeah, let's close it and consider this a transient bug in the kernel.

closed 2018-09-10 by greyltc

joserubenh, 2020-07-09:
I also have an e1000e  Intel I219-LM adapter.
I have the same symptoms listed above (dropping frames)
SimpleTest also gets spammed with ALL SLAVES RESUMED OPERATIONAL.
I also cannot write and read SDO.
SOEM works ok with realtek drivers. 
Maybe there is something in the firmware.

nakarlsson, 2020-07-11:
Assuming you’ve verified that the fix of this issue doesn’t help, write a new issue and post some info on your problem.