OpenEtherCATsociety/SOEM #179: does it support digital I/O mode?

meiguwiki, 2018-06-11:
does it support digital I/O mode without mcu?
If so, which test can I refer to?

nakarlsson, 2018-06-12:
If I understand you correct, Yes. The digital IO should include the PDO mappings in SII.

meiguwiki, 2018-06-12:
I only found this code to detect the slave pdi i/f type, and the Itype is not used in the whole SOEM
         context->slavelist[slave].Itype =
            etohs(ecx_APRDw(context->port, ADPh, ECT_REG_PDICTL, EC_TIMEOUTRET3)); /* read interface type of slave */

so I still need to add the pdo mapping by reading the SII in my testcase?

nakarlsson, 2018-06-12:
I guess you explicitly can read/write the ESC registers holding the digital I/O if you don’t want to do any PDO mapping.

meiguwiki, 2018-06-12:
I see, thanks

meiguwiki, 2018-06-13:
I found a xml to setup the pdo mapping for the digi I/O, and can I run with the simple_test?

				<Fmmu>Outputs</Fmmu>
				<Fmmu>Inputs</Fmmu>
				<Sm StartAddress="#x0f01" ControlByte="#x44" Enable="1" OpOnly="1">Outputs</Sm>
				<Sm StartAddress="#x1000" ControlByte="0" Enable="1">Inputs</Sm>
				<RxPdo Fixed="1" Mandatory="1" Sm="0">
					<Index>#x1600</Index>
					<Name>Output Process Data</Name>
					<Entry>
						<Index>#x7000</Index>
						<SubIndex>1</SubIndex>
						<BitLen>8</BitLen>
						<Name>Output</Name>
						<DataType>BITARR8</DataType>
					</Entry>
				</RxPdo>
				<TxPdo Fixed="1" Mandatory="1" Sm="1">
					<Index>#x1a00</Index>
					<Name>Input Process Data</Name>
					<Entry>
						<Index>#x6000</Index>
						<SubIndex>1</SubIndex>
						<BitLen>8</BitLen>
						<Name>Input</Name>
						<DataType>BITARR8</DataType>
					</Entry>
				</TxPdo>

meiguwiki, 2018-06-13:
actually my concern is, for the digital I/O mode, there is no MCU, 
so there is no mailbox in the slave configured.
does the SOEM can still read out the pdo mapping?
also which part of code does it as the Itype is never used, and how does the master know it's digital I/O without mailbox?

nakarlsson, 2018-06-13:
That information is/should  be provided by the slave, digital IO use the SII EEPROM for that. From there SOEM will get the PDO info needed to be able to configure the slave.

Mailbox / COE support or no support is found in the EEPROM.

You can use the SOEM test applications for digital IO.

mheden, 2018-06-13:
If you want the code, it is `ecx_map_sii` that reads the PDO mapping from the eeprom.

`ecx_config_find_mappings` first maps up all CoE slaves and then does the non-CoE.

meiguwiki, 2018-06-13:
@nakarlsson @mheden yep, I found these code today.
and I did run the test and see some reasonable traffic on wireshark
thank you
we can close it now

closed 2018-06-13 by nakarlsson