OpenEtherCATsociety/SOEM #180: Handling more than two groups

stefanpoeter, 2018-06-12:
Hi,

i am experiencing a weird behaviour when using more than two groups. 

````
    for (int i = 1; j <= groupcount; i += 1) {

        int wkc;
        int expectedWKC = (ec_group[i].outputsWKC * 2) + ec_group[i].inputsWKC;

            ec_send_processdata_group(i);

            wkc = ec_receive_processdata_group(i, EC_TIMEOUTRET);

            if (wkc != expectedWKC) {
                // mark group sync error
            }

````
When I use more than two groups, the working counter does not match the expected working counter for all groups except the first. `maxgroups` is set appropriately and the group ids in the slaves match my expectations. Any Idea what might be the problem?

nakarlsson, 2018-06-12:
Can you describe the setup? Groups and slaves per group.

Is the calculated value per group correct? Eg you receive a wkc value that is wrong?

stefanpoeter, 2018-06-12:
````
Setting slave 1 to group 1 (default)
Setting slave 2 to group 1 (default)
Setting slave 3 to group 1 (default)
Setting slave 4 to group 1 (default)
Setting slave 5 to group 1 (default)
Setting slave 6 to group 1 (default)
Setting slave 7 to group 2 (defaultB)
Setting slave 9 to group 2 (defaultB)
Setting slave 8 to group 3 (defaultC)
Setting slave 10 to group 3 (defaultC)
Setting slave 11 to group 4 (defaultD)
````

This is one possible setting, in general everything with more than two groups starts to have problems on the working counter returned from `ec_receive_processdata_group`. Could be that the first groups working counter is ok, but the rest isn't.

The slaves expected working counter:

````
Slave 1 expected working counter = 0
Slave 2 expected working counter = 1
Slave 3 expected working counter = 1
Slave 4 expected working counter = 2
Slave 5 expected working counter = 2
Slave 6 expected working counter = 1
Slave 7 expected working counter = 2
Slave 8 expected working counter = 1
Slave 9 expected working counter = 2
Slave 10 expected working counter = 3
Slave 11 expected working counter = 1
````

The groups expected working counter:

````
Group 1 expected working counter = 7
Group 2 expected working counter = 4
Group 3 expected working counter = 4
Group 4 expected working counter = 1
````

Here is the output from the missmatched working counter:

````
Group 2(defaultB) Working Count Missmatch (expected 4, actual = 6).
Group 3(defaultC) Working Count Missmatch (expected 4, actual = 8).
Group 3(defaultD) Working Count Missmatch (expected 8, actual = 1).
````

nakarlsson, 2018-06-12:
And of you compare expected and received wkc with wireshark?

stefanpoeter, 2018-06-13:
Haven't checked wireshark yet, will do that tomorrow.

Well the expected working counter are from the groups but I'll check the expected wkc from group 4 again tomorrow, should be 1, don't know why it says 8.

The values are calculated like this:

````
 int expectedWKC = (ec_group[i].outputsWKC * 2) + ec_group[i].inputsWKC;
````

stefanpoeter, 2018-06-14:
turns out I only set the maxgroup runtime variable but did not redefine EC_MAXGROUPS which is originally 2. It is working but I got a question: The way I see it, in order to work with a variable number of groups I need to fork your code and change the EC_MAXGROUPS in there. Is that intentional or do you have another way to do it?

nakarlsson, 2018-06-14:
The intention is to add a system to override it, I added issue that would address the possibillity to override settings. #65

closed 2018-06-14 by stefanpoeter