OpenEtherCATsociety/SOEM #181: slave groups with fmmus

meiguwiki, 2018-06-14:
As for the digital I/O test,
I have two fmmus configured as follows (from wireshark/eeprom)
fmmu0: log addr 0x00000000 (write only)
fmmu1: log addr 0x00000001 (read only)
The slave is not blockLRW as from eeprom, and it's ET1100 chip.

questions:
1. I found SOEM only tried to send LRW to 0x00000000 (for fmmu0 cfg) in the simple_test
  where is the frame to the address 0x00000001 ？

2. does the simple_test only test one group, group 0 (in my case it's output IO)?
   if so, I have to add the code for group1 for the input IO for the input IO?
 
 3. fmmu0 is configured as WR only, and should the datagram returns wkc with 2 or of 3 if SOEM sends LRW cmd? if so, why SOEM sends LRW but not LWR?

meiguwiki, 2018-06-14:
got you, The LRW will match the fmm0 for the write for the first byte/log address, and then match fmm1 for the read after one byte/log addr.
so the expected wkc is 2+1=3

nakarlsson, 2018-06-17:
Can we close this?

meiguwiki, 2018-06-17:
yes, I got it running after turn of the input IO as I didnt cfg the input correctly
output IO works as expected
please close it

closed 2018-06-17 by nakarlsson