OpenEtherCATsociety/SOEM #184: slave got lost on ubuntu16.04LTS

meiguwiki, 2018-06-23:
Hi, I have my simple test running with IO updates.
It ran well for tens of minutes, but randomly got slave lost afterwards.
I captured the log in wireshark, and see some weird things.
During suspicious range, 
the red LRW frame was sent in the first red rectangle，and didn't see the LRW frame back.
Several other ecat frames were sent afterwards, and then LRW is back with correct wkc in the second red rectangle. 
Everything looks ok in the wkc, and didnt see frame got lost.
However, the order of the frames seems got changed.

It looks like to me there is a buffer in the ethernet adapter interface on the master side?

BTW, I'm using ubuntu16.04LTS

ArthurKetels, 2018-06-25:
Did you check the coalesing and buffer settings of your NIC with ethtool?

meiguwiki, 2018-06-26:
these are my settings

Coalesce parameters for enp3s0:
Adaptive RX: off  TX: off
stats-block-usecs: 0
sample-interval: 0
pkt-rate-low: 0
pkt-rate-high: 0

rx-usecs: 1
rx-frames: 1
rx-usecs-irq: 0
rx-frames-irq: 5

tx-usecs: 1
tx-frames: 1
tx-usecs-irq: 0
tx-frames-irq: 5

rx-usecs-low: 0
rx-frame-low: 0
tx-usecs-low: 0
tx-frame-low: 0

rx-usecs-high: 0
rx-frame-high: 0
tx-usecs-high: 0
tx-frame-high: 0

meiguwiki, 2018-06-26:
I also noticed the SOEM doesnt retry on some BRD APRD FPRD frames
I guess the timeout of most (or all except process data frames??) ecat cmds should always be set to EC_TIMEOUTRET3 instead of EC_TIMEOUTRET

meiguwiki, 2018-06-30:
Thanks
it has been running for 2,000,000 cycles
looks like ok now, please close it

closed 2018-07-03 by nakarlsson

cklosters, 2019-11-22:
@meiguwiki, could you explain what resolved your issue. Changing your settings or EC_TIMEOUTRET? 

I am running into a similar issue working with 8 JVL MAC-00EC4 slaves wired in series and after a certain amount of time one of the slaves will be lost: Ubuntu 18.04 LTS