OpenEtherCATsociety/SOEM #185: Cannot manually configure PDO map

Stephenyu113, 2018-06-29:
Hi, I have a problem when configuring PDO map.
I try to control two Copley BEL drives through SOEM-master on Linux. But I can't use the xml file to make a PDO mapping, so I try to configure the PDO map manually using the method in issue #130.
https://github.com/hefloryd/SOEM/commit/22d8f5d
The thing is, there is a data type error when I use index 1c12 and 1c13, while I can scan my motor drives and make them OP.
I test the slaveinfo program, results are shown below:
------------------------------------------------------------------------
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Found BEL-090-14 at position 1
Found BEL-090-14 at position 2
Motor drive setup
Motor drive setup failed
Time:        8606.918 SDO slave:1 index:1c12.00 error:06070010 Data type does not match, length of service parameter does not match
Time:        8606.921 DSO slave:1 index:1c13.00 error:06070010 Data type does not match, length of service parameter does not match
Time:        8606.918 SDO slave:2 index:1c12.00 error:06070010 Data type does not match, length of service parameter does not match
Time:        8606.921 DSO slave:2 index:1c13.00 error:06070010 Data type does not match, length of service parameter does not match
2 slaves found and configured.
Calculated workcounter 0

Slave:1
Name:MOTOR
Output size: 0bits
Input size: 0bits
State: 4
Delay: 0[ns]
Has DC: 1
DCParentport: 0
Activeports: 1.1.0.0
Configured address: 1001
Man: 000000ab ID: 00001110 Rev: 00010003
SM0 A:1000 L: 256 F:00010026 Type:1
SM1 A:1100 L: 256 F:00010022 Type:2
SM2 A:1200 L: 256 F:00010024 Type:3
SM3 A:1800 L: 256 F:00010020 Type:4
FMMUfunc 0:1 1:2 2:0 3:0
MBX length wr: 256 rd: 256 MBX protocols : 0c
CoE details: 1d FoE details: 01 EoE details: 00 SoE details: 00
Ebus current: 0[mA]
only LRD/LWR: 0
PDO mapping according to CoE :
  SM2 outputs
        addr b    index:  sub  bitl  data_type    name
  SM3 intputs
        addr b    index:  sub  bitl  data_type    name

Slave2: .......

End slaveinfo, close socket
End program
-----------------------------------------------------------

I'd be appreciated for any response. Thanks so much!

nakarlsson, 2018-07-03:
To clarify, when you configure them using the default and you can reach OP, CoE is not used? Instead the mapping is in the EEPROM?

Stephenyu113, 2018-07-03:
Thanks for your reply.
The servo drive provides a xml file which maps all the SDOs and PDOs, I must put it into TwinCAT to configure the mapping. So, I think if I can't apply this xml file, the drive will tell nothing about PDO mapping in the EEPROM.
But I tried the EK1100 and EL3064, they really showed some default CoE mapping. So I think it's due to the servo drive itself that I can't acquire PDO mapping without xml?

nakarlsson, 2018-07-03:
Can you post the XML? In Zip.

Stephenyu113, 2018-07-03:
[Copley_BEL_1.68.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2160655/Copley_BEL_1.68.zip)

ArthurKetels, 2018-07-03:
Copley drives are technically nice, but the EtherCAT interface is clearly not. They just converted a CanOpen drive to EtherCAT without using any of the nice features of EtherCAT. What is new...

I do not have code for a BEL drive but below is a code snippet for a Copley AEP drive. And, yes, this is pascal.

```pascal
procedure AEPsetup(slave : UINT16); cdecl;
var u8val : uint8;
    u16val : uint16;
begin
    u8val := 0;
    ec_SDOwrite(slave, $1c12, $00, FALSE, sizeof(u8val), @u8val, EC_TIMEOUTRXM);
    u16val := $1603;
    ec_SDOwrite(slave, $1c12, $01, FALSE, sizeof(u16val), @u16val, EC_TIMEOUTRXM);
    u8val := 1;
    ec_SDOwrite(slave, $1c12, $00, FALSE, sizeof(u8val), @u8val, EC_TIMEOUTRXM);

    u8val := 0;
    ec_SDOwrite(slave, $1c13, $00, FALSE, sizeof(u8val), @u8val, EC_TIMEOUTRXM);
    u16val := $1a02; // $1a02 actual position $1a03 actual velocity
    ec_SDOwrite(slave, $1c13, $01, FALSE, sizeof(u16val), @u16val, EC_TIMEOUTRXM);
    u8val := 1;
    ec_SDOwrite(slave, $1c13, $00, FALSE, sizeof(u8val), @u8val, EC_TIMEOUTRXM);

    u8val := 3;
    ec_SDOwrite(slave, $6060, $00, FALSE, sizeof(u8val), @u8val, EC_TIMEOUTRXM);

    u16val := 30;
    ec_SDOwrite(slave, $2300, $00, FALSE, sizeof(u16val), @u16val, EC_TIMEOUTRXM);

    u16val := 2498;
    ec_FPWR(pec_slave^[slave].configadr, $0400, sizeof(u16val), @u16val, EC_TIMEOUTRET);
    u16val := 1000;
    ec_FPWR(pec_slave^[slave].configadr, $0410, sizeof(u16val), @u16val, EC_TIMEOUTRET);
    u16val := 1000;
    ec_FPWR(pec_slave^[slave].configadr, $0420, sizeof(u16val), @u16val, EC_TIMEOUTRET);

end;
```

ArthurKetels, 2018-07-03:
>SDO slave:1 index:1c12.00 error:06070010 Data type does not match, length of service parameter does not match

Sometimes those indexes have subindex 0  internally defined as uint16 even when the xml says otherwise. Just try uint16 for subindex 0

Stephenyu113, 2018-07-03:
Thanks, I've tried uint16 for 1c12, and I think I used ec_SDOwrite function as you wrote here. But I don't know why there's a data type error. And I tried selective pdo mapping method as same as the example. https://github.com/hefloryd/SOEM/commit/22d8f5d . I thought it gonna be correct, but error was still there.

ArthurKetels, 2018-07-03:
Can you capture the configuration phase with wireshark and zip the .pcap file here? Perhaps I can find something there.

Stephenyu113, 2018-07-03:
[Copley_Wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2161231/Copley_Wireshark.zip)

nakarlsson, 2018-07-04:
I think there is a bug in hefloryd/SOEM@22d8f5d

The final write to 1c12/1c13 subindex 0 is write16, should be write8, you can see the first write succeed in your wireshark log, it is 1Byte, the second write that failes is 2 Bytes.

Stephenyu113, 2018-07-04:
Thanks so much for you both. Things are all set. So I close the issue.
Hope your guideline also helps others.

closed 2018-07-04 by Stephenyu113