OpenEtherCATsociety/SOEM #188: simple_test cant change state to operational , status dont show any error

Katod, 2018-07-25:
My test configuration Beckhoff EK1100 EL1008 EL2008 , when run simple test , have message about Not all slaves reached operational state.

> ./simple_test enp30s0
> SOEM (Simple Open EtherCAT Master)                                                                                                                                                                                                                                             
> Simple test                                                                                                                                                                                                                                                                    
> Starting simple test                                                                                                                                                                                                                                                           
> ec_init on enp30s0 succeeded.                                                                                                                                                                                                                                                  
> 3 slaves found and configured.                                                                                                                                                                                                                                                 
> Slaves mapped, state to SAFE_OP.                                                                                                                                                                                                                                               
> segments : 1 : 2 0 0 0                                                                                                                                                                                                                                                         
> Request operational state for all slaves                                                                                                                                                                                                                                       
> Calculated workcounter 3                                                                                                                                                                                                                                                       
> Not all slaves reached operational state.
> Slave 1 State=0x01 StatusCode=0x0000 : No error
> Slave 2 State=0x01 StatusCode=0x0000 : No error
> Slave 3 State=0x01 StatusCode=0x0000 : No error
> 
> Request init state for all slaves
> End simple test, close socket
> End program

Status dont have any error , output in ec_slave can change but change state cant.
Its slave info with map parameter

_

> SOEM (Simple Open EtherCAT Master)
> Slaveinfo
> Starting slaveinfo
> ec_init on enp30s0 succeeded.
> 3 slaves found and configured.
> Calculated workcounter 3
> Not all slaves reached safe operational state.
> Slave 1 State= 1 StatusCode=   0 : No error
> Slave 2 State= 1 StatusCode=   0 : No error
> Slave 3 State= 1 StatusCode=   0 : No error
> 
> Slave:1
>  Name:EK1100
>  Output size: 0bits
>  Input size: 0bits
>  State: 1
>  Delay: 0[ns]
>  Has DC: 1
>  DCParentport:0
>  Activeports:1.1.0.0
>  Configured address: 1001
>  Man: 00000002 ID: 044c2c52 Rev: 00120000
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: -2000[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
> 
> Slave:2
>  Name:EL1008
>  Output size: 0bits
>  Input size: 8bits
>  State: 1
>  Delay: 145[ns]
>  Has DC: 1
>  DCParentport:1
>  Activeports:1.1.0.0
>  Configured address: 1002
>  Man: 00000002 ID: 03f03052 Rev: 00110000
>  SM0 A:1000 L:   1 F:00010000 Type:4
>  FMMU0 Ls:00000001 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:01 Act:01
>  FMMUfunc 0:2 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 90[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
>   SM0 TXPDO 0x1A00 Channel 1
>      addr b   index: sub bitl data_type    name
>   [0x0001.0] 0x6000:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A01 Channel 2
>      addr b   index: sub bitl data_type    name
>   [0x0001.1] 0x6010:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A02 Channel 3
>      addr b   index: sub bitl data_type    name
>   [0x0001.2] 0x6020:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A03 Channel 4
>      addr b   index: sub bitl data_type    name
>   [0x0001.3] 0x6030:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A04 Channel 5
>      addr b   index: sub bitl data_type    name
>   [0x0001.4] 0x6040:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A05 Channel 6
>      addr b   index: sub bitl data_type    name
>   [0x0001.5] 0x6050:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A06 Channel 7
>      addr b   index: sub bitl data_type    name
>   [0x0001.6] 0x6060:0x01 0x01 BOOLEAN      Input
>   SM0 TXPDO 0x1A07 Channel 8
>      addr b   index: sub bitl data_type    name
>   [0x0001.7] 0x6070:0x01 0x01 BOOLEAN      Input
> 
> Slave:3
>  Name:EL2008
>  Output size: 8bits
>  Input size: 0bits
>  State: 1
>  Delay: 300[ns]
>  Has DC: 1
>  DCParentport:1
>  Activeports:1.0.0.0
>  Configured address: 1003
>  Man: 00000002 ID: 07d83052 Rev: 00100000
>  SM0 A:0f00 L:   1 F:00090044 Type:3
>  FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
>  FMMUfunc 0:1 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 110[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
>   SM0 RXPDO 0x1600 Channel 1
>      addr b   index: sub bitl data_type    name
>   [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1601 Channel 2
>      addr b   index: sub bitl data_type    name
>   [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1602 Channel 3
>      addr b   index: sub bitl data_type    name
>   [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1603 Channel 4
>      addr b   index: sub bitl data_type    name
>   [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1604 Channel 5
>      addr b   index: sub bitl data_type    name
>   [0x0000.4] 0x7040:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1605 Channel 6
>      addr b   index: sub bitl data_type    name
>   [0x0000.5] 0x7050:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1606 Channel 7
>      addr b   index: sub bitl data_type    name
>   [0x0000.6] 0x7060:0x01 0x01 BOOLEAN      Output
>   SM0 RXPDO 0x1607 Channel 8
>      addr b   index: sub bitl data_type    name
>   [0x0000.7] 0x7070:0x01 0x01 BOOLEAN      Output
> End slaveinfo, close socket
> End program

_

maybe someone have problem like this , thanks.

Katod, 2018-07-30:
Have false read access EEprom but when use eepromtool for write and read EEprom all good.
```
Frame 56: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface 0
Ethernet II, Src: 03:01:01:01:01:01 (03:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
    Source: 03:01:01:01:01:01 (03:01:01:01:01:01)
    Type: EtherCAT frame (0x88a4)
EtherCAT frame header
    .... .000 0001 0010 = Length: 0x012
    .... 0... .... .... = Reserved: Valid (0x0)
    0001 .... .... .... = Type: EtherCAT command (0x1)
EtherCAT datagram(s): 'FPWR': Len: 6, Adp 0x1001, Ado 0x502, Wc 1 
    EtherCAT datagram: Cmd: 'FPWR' (5), Len: 6, Adp 0x1001, Ado 0x502, Cnt 1
        Header
            Cmd        : 5 (Configured address Physical Write)
            Index: 0x0a
            Slave Addr: 0x1001
            Offset Addr: 0x0502
            Length     : 6 (0x6) - No Roundtrip - Last Sub Command
            Interrupt: 0x0004
        EEPROM Ctrl/Status (0x502): 0x0100, Read access
            .... ...0 = Write access: False
            ..0. .... = EEPROM emulation: Normal operation
            .0.. .... = 8 byte access: False
            0... .... = 2 byte address: False
            .... .... = Read access: True
            .... .... = Write access: False
            .... .... = Reload access: False
            .... .... = CRC error: False
            .... .... = Load error: False
            .... .... = Cmd error: False
            .... .... = Write error: False
            .... .... = Busy: False
        EEPROM Address Lo (0x504): 0x0008
        EEPROM Address Hi (0x506): 0x0000
        Working Cnt: 1
Pad bytes: 000000000000000000000000000000000000000000000000...

```
trying find how fix this and turn on write access but dont find info , maybe some one know and have similar problem

ArthurKetels, 2018-08-17:
This frame does not show the problem you have. It is just initialisation at start-up to get into a know state. Please attach a full wireshark capture (zipped) from the start to end of initialisation.

closed 2018-09-03 by Katod