OpenEtherCATsociety/SOEM #190: Configuration using ESI file

dotdotdotdash, 2018-07-27:
I'm currently building a drive system for my elmo gold series drives. I was able to get my drive system going with soem and by manually configuring the drives individually using elmo application studio. I read through certain issues here that soem doesn't support configuration setup through esi/ eni. I was just wondering what could be the best way of doing it if I want to embark upon that route ? 

Thanks !

nakarlsson, 2018-07-28:
ESI, Add an 3ed config possibillity, currently SOEM use CoE & SII in that order, done in config init.

Add an XML parser abstraction layer for that configuration option, I think different systems might wanna use different parsers depending on size.

nakarlsson, 2018-09-28:
@dotdotdotdash, are you looking into this? It is an interesting topic but I think a first step would be to handle ENI files rather than ESI. Creating ENI files from ESI could be a side project in any language.

nakarlsson, 2018-12-18:
Might add support to start from ENI in the future, rather than ESI

closed 2018-12-18 by nakarlsson