OpenEtherCATsociety/SOEM #191: Can the update-rate / performance be increased

nvandenbraber, 2018-08-03:
Hello all, I am investigating the option of using SOEM in the real-time control of our robot. I have SOEM running on a my system using a Linux preempt-rt kernel with an Intel core i7 CPU and Intel pro1000 card. My current performance in terms of update-rate is not that great, I can get it to run around 1-2 Khz. Does anybody have any experience with using other ethernet cards or drivers? Or is there any other way to speed up / tweak SOEM to increase the performance? All tips and tricks are welcome.

nakarlsson, 2018-08-03:
See if you can get any ideas from https://github.com/OpenEtherCATsociety/SOEM/issues/171.

nvandenbraber, 2018-08-06:
Hi nakarlsson, thank you for pointing me to toward that direction. I will start with tuning the e1000 driver with ethtool to see how it performs. At a first glance it does make a huge difference. Thanks.

nvandenbraber, 2018-08-13:
Solved, i successfully tuned the driver with ethtool.

nakarlsson, 2018-08-13:
Good, I’ll close the issue

closed 2018-08-13 by nakarlsson