OpenEtherCATsociety/SOEM #192: SOEM can't output data with simple_test

TroyChu, 2018-08-06:
A few days ago, my SOEM works well, but today when I try the simple_test, it can not output data. It will either keep outputting "OK : all slaves resumed OPERATIONAL" or stuck at "Calculated workcounter 3". What wrong with the SOEM? What should I do?
[image]
[image]

ArthurKetels, 2018-08-17:
The last screen shows slave state 0x14 (Pre operational + error) and the error status code is 0x001a (Sync error). Clearly your timing is not set-up correctly for your slave. Check the slave requirements and compare against the real timing as captured by wireshark.

Some slaves for example require a constant PDO update with fixed time interval during the PreOp To OP transition. Some even require the maximum deviation to be within strict bounds.

nakarlsson, 2018-09-10:
@TroyChu , can we close this issue?

nakarlsson, 2018-09-28:
We'll close

closed 2018-09-28 by nakarlsson