OpenEtherCATsociety/SOEM #195: Example code for reading/writing OD entries, PDO mapping and sending out RPDOs?(CoE)

mohse-n, 2018-08-10:
Hi. I'm looking for code for these scenarios:
1- **Reading/writing OD entries (using SDOs)**: As an example, how can I change the value of the entry at 0x6040 (controlword) to 6?
2- **PDO mapping**: How can I set 0x60C1sub1 (interpolation data record) as a RPDO?
3- **Sending out RPDO**: How can I specify a value for and send RPDO1?
Note that I'm familiar with the underlying protocol (CAN in Automation) and just need some examples of using SOEM'S API on Linux.

nakarlsson, 2018-08-10:
You’ll find such examples in the test folders, ex https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/linux/ebox/ebox.c

mohse-n, 2018-08-11:
Thanks. I encountered the following line in `ebox.c`:
`ec_SDOwrite(1,0x1c12,01,FALSE,os,&ob2,EC_TIMEOUTRXM);`
While the definition of `ec_SDOwrite` specifies its first input as a `ecx_contextt*` type. I would like to know why this input is ignored in the line above and whether that's usually the case or not.

mheden, 2018-08-11:
All ec_ functions are using a static allocated structure. Functions
prefixed ecx_ are using your own allocated ecx_contextt structure.

mohse-n, 2018-08-12:
One thing that's still unclear to me: how do we send PDOs? Suppose that I have mapped the entry 0x60C1sub1 as RxPDO1 and some other entry as RxPDO2. How can I:
1- Assign a value to their corresponding position in the EtherCAT frame?
The function that I think might be related to this issue is` ec_RxPDO`. But I haven't seen it used in any of the example codes. Am I on the right track?
2- Send out the EtherCAT frame?
I have no idea about this one.

nakarlsson, 2018-08-12:
1. The IOmap is the local memory used by send/recv processdata 
2. The slave structure have a ref to slaves input/outputs mapping in the IOmap
3. You can ref the slave ref accordingly, in ebox.c check ex usage of out_EBOX.
4. send/recv_processadata in the same ebox.c do the actual send/recv on the EtherCAT bus.

mohse-n, 2018-08-12:
Thanks. I think have obtained the clues that I needed.

closed 2018-08-12 by mohse-n

vmc-7645, 2025-03-04:
Is there an example piece of PDO actively being utilized? Thus far I've only found code in which it is supposedly activated (i.e. [earlier reply reference](https://github.com/OpenEtherCATsociety/SOEM/issues/195#issuecomment-412093533)), and never actually used via write (ec_RxPDO) or read (ec_TxPDO).

ArthurKetels, 2025-03-05:
Please do not hijack old posts. Open a new one if you have questions. 

To answer this quickly, there are no slave implementations that support ec_TxPDO or ec_RxPDO via mailbox communication. Use the synchronous PDO (LRW, LRD, LWR) instead .