OpenEtherCATsociety/SOEM #196: Basic questions about IOmap

mohse-n, 2018-08-12:
Hi. I have some questions regarding the variable IOmap, as defined in test codes:
1- How does it relate to an EtherCAT frame?(or EtherCAT standard in general)
2- What affects its size? For example, in most example codes I see it defined as`char IOmap[4096];` Why the 4096-char length?

nakarlsson, 2018-08-12:
1. Process data from RXPDO, TXPDO objects. The IOmap layout is calculated and created by SOEM. You might get an idea by reading https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html.
2. It is sent as payload in an EtherCAT datagram by using the logic read/write commands. An EtherCAT frame have 1-N datagrams.
3. The size is the sum of active/selected RXPDO/TXPDOs, 4096 is just a very big or small IOmap. As long as it cover your RXPDO/TXPDOs your good. Ex having an EL1008 of 8bits would be covered by a char IOmap[1].

Note: SOEM have 2 different ways to create the IOmap layout that affect the size. The legacy is described in the tutorial. The other can be find in the following PR, https://github.com/OpenEtherCATsociety/SOEM/pull/156.

mohse-n, 2018-08-15:
Thanks. I will close this issue for now.

closed 2018-08-15 by mohse-n